Asuswrt-Merlin/asuswrt-merlin.legacy #922: Traditional QoS broken on SDK7 builds

RMerl, 2016-06-05:
Traditional QoS does not work properly on newer SDK7 models.

Issue is known to Asus, but there is no ETA as to when it will be resolved.

There's a good chance the root issue lies in the closed source code (possibly the Ethernet driver, or the GMAC3 architecture), so it cannot be fixed by outside developers.

klaxian, 2016-09-23:
This also affects my RT-AC68U. No response from Asus.

adrianmace, 2016-12-11:
RT-AC87U checking in with this issue.

donattila, 2016-12-14:
There should be a warning when enabling QoS (In the meantime while its not working as supposed). I searched about a month why my WAN-Speed was only 5 Mbits instead of 100, so the last thing I did after checking all the hardware inside my network I disabled QoS and it was fast again. I already thought my ISP is the fault :P

rkkoszewski, 2017-09-02:
Any news on this? I have a RT-AC68U, and I was looking forward to use this feature, to prevent Guests from using up the whole internet when they are downloading. Adaptive QoS doesn't seem to support a per-client configuration, and when somebody is downloading a Game trough Steam, the whole network is barely usable. Hope Asus will get this fixed some day... I feel scammed. At least bandwidth limiting works, but its far from ideal.

oldjamey, 2017-09-30:
I just opened a support case with Asus and tried to report this bug and a simple fix to workaround it, which should also be able to be easily applied to the Merlin firmware.  It's a really simple issue with a really simple workaround.  The download speed and upload speed text area labels under Traditional QoS have gotten flip-flopped.

I had put up with crappy Internet for far too long and decided to dig into the root cause.  I took the router out of the situation, because I too thought it was my ISP.  Then I realized my speed from the ISP was just fine, so it had to be my QoS settings.  I remembered that somewhere in the official firmware releases, the Upload Speed became the label on top instead of at the bottom on the Traditional QoS screen (the screen that says to enter your speeds from your ISP or speedtest.net).

On a whim, I put my upload speed in the download speed textarea and my download speed in the upload speed textarea, applied the settings, and BOOM!  Traditional QoS fixed.  Bingo-bango.
I double I will get credit anywhere for this, but I want as many people as possible to know about it in order to improve customer experience.  I hope this helps.

rkkoszewski, 2017-09-30:
Hi, I found that out too a while ago. But that only fixes download/upload speed. But from what I can see, QOS is still not working. I have a user (from a guest network) downloading something trough Steam using up the whole internet. I enabled traditional QoS and gave him the lowest priority, setting all the parameters to ridiculous amounts like 5 to 10%, and me (main WiFi) I set the highest priority. Still I get barely 1 mb of bandwidth while the guest is using 20 mb, making my internet basically unusable with pages sometimes even timing out. Didn't test this with two computers in the main WiFi. Somebody has to dig deeper into this. Swapping upload and download sure does something, but doesn't fix the non working QoS from what I could observe.

void12345, 2017-10-08:
@oldjamey I tried your method, still not working (i'm using merlin)

rkkoszewski, 2017-12-17:
Any news on this? I updated to the 380.69 FW and the issue still persists. I specifically bought this router to  balance traffic between the connected computers in my network (Web downloads). Is there any workaround to balance the traffic by device similar to Traditional QOS? 

Is there any place to check if anything is done about this? How long has it been that this is not working?

oldjamey, 2017-12-28:
Just to update this thread, I am back to using the latest stock Asus firmware.  Still have my upload and download speeds reversed, and I am blazing along perfectly.  Traditional QoS is working as expected, no problems whatsoever.

If this part of the Merlin build is still based off the open source stock firmware, then I think you guys need to give my previous insight a little more credit.  I can't be a lone wolf here, and I'm not special.  Using Charter Spectrum cable with like 6-7Mbps upload on a good day.  That's currently in my download speed text area, with my max download speed in my upload speed textarea.

It may sound goofy and like I'm doing some sort of rain dance and have no idea what I'm talking about...but it works.  Plain and simple.

rkkoszewski, 2017-12-28:
Have you tried if it really works? I mean the QoS. Having one computer set as high priority and another as low priority. Set the low priority to have a max of 10% bandwidth and the high one to 90%. Then start downloading trough Steam on both PC's and see if high priority one gets the biggest chunk of bandwidth. If it works then I guess you're right.

I tried that out on a previous build, and it did not work for me. If lucky I get a 50/50 bandwidth usage on both PC's but the same happens without any QoS. Try it out and let us know.