Asuswrt-Merlin/asuswrt-merlin.legacy #923: MIPS builds failing due to missing closed source components

RMerl, 2016-06-05:
Asus's 380_3264 GPL release for the N66 and AC66 are missing the new shared/prebuild/private.o binary blob, so building of the two MIPS models will fail.

mstombs, 2016-07-12:
New GPL sourcecode now released, and it includes shared/prebuilt/private.o and tcode.o

Description GPL of ASUS RT-N66U for firmware 3.0.0.4.380.3831
File Size 561.6 MBytes 2016/07/11 update

http://dlcdnet.asus.com/pub/ASUS/wireless/RT-N66U_B1/GPL_RT_N66U_30043803831.zip

RMerl, 2016-07-12:
Yes, but I can't use those closed source components as they are specific to the new GPL.  So I will first need to merge the new GPL code, test that new code, and also obtain the closed source components for every other models (so far, only N66, AC66 and AC68 are available).

Overall, this won't happen for a few weeks.

RMerl, 2016-07-12:
Since 380.60's GPL code isn't stable enough (a few hard-to-track WAN issues for some users), the current plan is to abandon 380.60 development and leave it as a beta-only release, and start working on 380.61 based on the newer GPL code.  That way, for the time being people with MIPS models will get "post-new-validation" support with 380.61, and ARM users will get it from 380.60.  Until Asus (hopefully) provides the necessary components for ARM models with the new 380_38xx GPL code, so everyone can be brought together on that new codebase.

mstombs, 2016-07-13:
Great, thanks, due to the firmware validation issue the next release is a major step and worth waiting for, after your summer holidays!  Can only hope Asus don't add more models, bump version No and make the new MIPS private.o 'out of date'  when updating all the ARM firmwares.

RMerl, 2016-07-31:
Fixed with commit bcde2858bf296d645409a3c7d9014b210c7e9199 .

closed 2016-07-31 by RMerl