Asuswrt-Merlin/asuswrt-merlin.legacy #924: VPN client blocks FTP and VPN connections

jhallenb, 2016-06-06:
When enable the router to act as a VPN client so all computers on the LAN goes through the VPN connection (PPTP), then accessing the router's share over FTP is blocked. Also if one enable the router to act also as OpenVPN server it is not possible to have computers connect to the router over VPN.

Seems like a bug, all incomming connections like FTP and OpenVPN seems to routed out over the PPTP tunnel hence does no longer work.

RMerl, 2016-06-06:
Asus recently tightened security so that various daemons will now only listens for connections to the LAN's subnet.  Is your VPN within the same subnet as your LAN, or a different one?

jhallenb, 2016-06-06:
Router is at 192.168.1.1
Router is connected to ISP using direct cable.
I am using a VPN provider, only PPTP, and enable a tunnel to the VPN provider using the PPTP client in the router. So all LAN computers are reaching internet through the tunnel. 

In this configuration nobode outside my LAN can reach the USB share over FTP.
If I setup an OpenVPN server in the router I cannot connect to it from outside my LAN as long as the PPTP tunnel is activated. The OpenVPN server I usually setup to be on 192.168.2.1/24

RMerl, 2016-06-06:
This is a known technical limitation.  If you have a VPN client set to forward all traffic through the tunnel, then inbound connection to the router won't work properly due to a routing conflict - it will try to forward the outbound traffic through the tunnel rather than through the WAN interface.

Proper workaround is to use OpenVPN instead of PPTP, and setup policy-based routing, with a rule forcing all traffic to the router's IP to go through the WAN interface rather than the tunnel.

jhallenb, 2016-06-06:
Okay, a known technical limitation, but could it be resolved for PPTP?

For OpenVPN and policy routing, wouldn't I need to know the public IP to the router then? My ISP provides a dynamic IP...so that is not a solution unless one can anable this automatically for the current public WAN IP. I have seen no way to solve this, but maybe I have missed something?

RMerl, 2016-06-08:
Please head up to the support forums if you need help.

closed 2016-06-08 by RMerl