Asuswrt-Merlin/asuswrt-merlin.legacy #925: Ignore generated samba file

tuapuikia, 2016-06-07:
Hi Eric,

I think the following path should be ignored during commit, the files will be generate during configure and compile

../release/src/router/samba36/source3/librpc/gen_ndr/

RMerl, 2016-06-08:
They only get generated if they don't exist, and the auto-generated ones fail to compile for some reason.

OpenWRT also includes precompiled (and patched) copies of these, so I assume they have the same problem.

tuapuikia, 2016-06-09:
Compile failed when using the committed files.

Compiling ../libcli/security/dom_sid.c
librpc/gen_ndr/ndr_dcerpc.c: In function ‘ndr_push_dcerpc_sec_vt’:
librpc/gen_ndr/ndr_dcerpc.c:3482:31: error: ‘LIBNDR_FLAG_SUBCONTEXT_NO_UNREAD_BYTES’ undeclared (first use in this function)
librpc/gen_ndr/ndr_dcerpc.c:3482:31: note: each undeclared identifier is reported only once for each function it appears in
librpc/gen_ndr/ndr_dcerpc.c: In function ‘ndr_pull_dcerpc_sec_vt’:
librpc/gen_ndr/ndr_dcerpc.c:3506:31: error: ‘LIBNDR_FLAG_SUBCONTEXT_NO_UNREAD_BYTES’ undeclared (first use in this function)
The following command failed:
arm-brcm-linux-uclibcgnueabi-gcc -DLINUX26 -DCONFIG_BCMWL5 -DDEBUG_NOISY -DDEBUG_RCTEST -pipe -DBCMWPA2 -DBCMARM -fno-strict-aliasing -marm -DRTCONFIG_NVRAM_64K -DLINUX_KERNEL_VERSION=132644 -O3 -ffunction-s
ections -fdata-sections -I/root/asuswrt-phantom/release/src-rt-6.x.4708/router/libiconv-1.14/include -DBCMARM -I. -I/root/asuswrt-phantom/release/src/router/samba36/source3 -I/root/asuswrt-phantom/release/sr
c/router/samba36/source3/../lib/popt -I/root/asuswrt-phantom/release/src/router/samba36/source3/../lib/iniparser/src -Iinclude -I./include  -I. -I. -I./../lib/replace -I./librpc -I./.. -I./../lib/talloc -I./
../lib/tevent -I../lib/tdb/include -DHAVE_CONFIG_H  -DNDEBUG -DSHMEM_SIZE=524288 -Dfcntl=fcntl64 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE=1 -D_LARGEFILE64_SOURCE=1 -D_LARGE_FILES=1 -Iinclude -I./include -I
. -I. -I./../lib/replace -I./librpc -I./.. -I./../lib/popt -I/root/asuswrt-phantom/release/src-rt-6.x.4708/router/libiconv-1.14/include  -I/root/asuswrt-phantom/release/src/router/samba36/source3/lib -I.. -D
_SAMBA_BUILD_=3 -D_SAMBA_BUILD_=3 -fPIC -c librpc/gen_ndr/ndr_dcerpc.c -o librpc/gen_ndr/ndr_dcerpc.o
Compiling ../libcli/security/security_descriptor.c
make[5]: *** [librpc/gen_ndr/ndr_dcerpc.o] Error 1
make[5]: *** Waiting for unfinished jobs....
Compiling ../libcli/security/security_token.c
make[5]: Leaving directory `/root/asuswrt-phantom/release/src/router/samba36/source3'
make[4]: *** [apps] Error 2
make[4]: Leaving directory `/root/asuswrt-phantom/release/src/router/samba36'
make[3]: *** [samba36] Error 2

2 possible solution,

1. git checkout b0fcf69b5b859faf31dc7bc2c7d8b0d5610a619e release/src/router/samba36/source3/librpc/gen_ndr/dcerpc.h release/src/router/samba36/source3/librpc/gen_ndr/ndr_dcerpc.c release/src/router/samba36/source3/librpc/gen_ndr/ndr_dcerpc.h release/src/router/samba36/source3/librpc/gen_ndr/py_dcerpc.c

2. rm -v release/src/router/samba36/source3/librpc/gen_ndr/dcerpc.h release/src/router/samba36/source3/librpc/gen_ndr/ndr_dcerpc.c release/src/router/samba36/source3/librpc/gen_ndr/ndr_dcerpc.h release/src/router/samba36/source3/librpc/gen_ndr/py_dcerpc.c
and build with cleanup option to regenerate gen_ndr file.

RMerl, 2016-06-09:
It builds fine for me and for everyone else at SNBForums doing their own personal builds.

tuapuikia, 2016-06-13:
Seem like I have to clean up before rebuilt. Not a big issue.

closed 2016-06-13 by RMerl