Asuswrt-Merlin/asuswrt-merlin.legacy #926: Upload speed capped when Merlin was installed

inquam, 2016-06-07:
I have Fiber in my house and had my Asus ac68u connected to the fiber box. With the stock firmware I got about 245MBit/sec down and 247MBit/sec up. Since my subscription is for 250/250 it's quite close.

I then installed Merlin 380.59 and after that I still get around 240MBit/sec down, but my up speed seems to be stuck at ~145MBit/sec.

If I connect a computer directly to the fiber box I get my original speeds back, so the issue seems to be with the router and it started when I installed Merlin. Any ideas on what the cause could be?

RMerl, 2016-06-07:
Please use the forums for support questions, this tracker is for bug reports.

closed 2016-06-07 by RMerl

inquam, 2016-06-07:
Ahhh sorry... But I have not been able to find a forum :)

stevek123, 2016-06-07:
http://www.snbforums.com/forums/asuswrt-merlin.42/