Asuswrt-Merlin/asuswrt-merlin.legacy #929: No way to change Language

omniwired, 2016-06-08:
[image]

RMerl, 2016-06-09:
I've seen someone report the same problem with Asus's stock firmware, and he was also using Japanese.  I suspect it's related to that, as the language selector appears normally with other languages.

RMerl, 2016-06-09:
Asus is actually doing it on purpose, hiding the language selector for the JP region:

https://github.com/RMerl/asuswrt-merlin/blob/master/release/src/router/httpd/web.c#L6687

Anyone knows why that might be the case?  Is there some odd Japanese law enforcing that kind of behavior?

omniwired, 2016-06-09:
maybe to aid their uses, there are no laws banning to use english in products....

Also this router came with multi lingual UI.

omniwired, 2016-06-09:
Thanks so much for your help and your amazing work in Merlin, I really love it.

For the time being I will try to downgrade the firmware to 380.57 that had English support.

Maybe we can just get rid of that IF, since there isn't a good reason that is happening.

I could make a PR if you wish.

RMerl, 2016-06-09:
Thanks, but I'll just give it some time to see if anyone can come up with a good explanation, otherwise I will take care of it.

closed 2016-06-10 by RMerl

omniwired, 2016-06-14:
saw your changes, thanks for the fix! :+1:

songxinjing, 2016-06-15:
I have already flashed version 380.60 alpha 3, and I have saw the language selector, thanks for your work. **But**, I haven't changed the language, If I change the language to English or others, it remains Japanese.

RMerl, 2016-06-15:
Found additional code where Asus were enforcing Japanese language for routers with a JP region code.

nudls, 2017-03-26:
Hello, I've upgraded my rt-ac87u to latest firmware and now all gui in japanese. Applying RT-AC87U_380.66_alpha2 will help me get english back?
thanks

commit 567f226a (2016-06-10): httpd: Do not remove the language selector if the router's region is JP (fixes #929)