Asuswrt-Merlin/asuswrt-merlin.legacy #934: Hide passwords by default on the VPN server pages

RMerl, 2016-06-10:
If you configure any users on the VPN server pages, these are shown by default, which can be a security problem if you just want to access that page to create a user while someone else is watching.

Hide the passwords by default, with a checkbox allowing to reveal it, a bit similar to what is used for the router password on the Administration page.

pepeEL, 2016-08-02:
Any plan when will be this add ?

RMerl, 2016-08-02:
Not at this time, no.

pepeEL, 2016-08-02:
Why :) ?

RMerl, 2016-08-02:
Because there's one single developer who already has to do **everything** on his own.  So at this time, I don't know when it will be added.  Time is a limited resource, and I have a personal life outside of this project.

pepeEL, 2016-08-02:
Ahh Ok sorry :) But thanks for all you good job :)

commit 30ab1bee (2016-08-25): webui: move OpenVPN server passwords to an array instead of relying on DOM storage, and add a checkbox to reveal/hide passwords from the table (by deaults, passwords are no longer shown).  Closes #934.

Also cleaned up some of the HTML whitespace.
closed 2016-08-25 by RMerl