Asuswrt-Merlin/asuswrt-merlin.legacy #935: Assign VLAN tag to guest network

ax42, 2016-06-10:
For example the airport express apparently does that, and it means that a gateway router can then isolate the guest network when the Asus is only being used as an AP.

http://community.ubnt.com/t5/EdgeMAX/Apple-Airport-amp-Guest-WiFi-network/td-p/787204 for an example discussion.

RMerl, 2016-06-10:
Such architectural changes would be beyond the scope of this project, and make it difficult to keep the codebase in sync with Asus's own code.

ax42, 2016-06-11:
No problem, thanks -- I had no idea how complicated it would be, so I thought I'd open the issue and find out.

closed 2016-06-11 by ax42

RMerl, 2016-06-11:
Note that Asus started implementing VLAN support a few months ago, but haven't finished doing so yet.

pepeEL, 2016-06-11:
From what you know that ASUS start implement it ?

RMerl, 2016-06-11:
No idea.  The only thing I know is what I saw them add to the source code, no idea if/when that would get finalized.  Sometimes they start working on something and never finish it (like the Tor support, or the Norton gateway security support).