Asuswrt-Merlin/asuswrt-merlin.legacy #936: Asrurs RT-n66u, 380.59, broken IPv6 support

monosoul, 2016-06-13:
Hi!
After update to 380.59 IPv6 stopped working (router doesn't receiving IPv6 setting from ISP). With 380.58 everything is OK.

RMerl, 2016-06-13:
Works for other users, so please head to the support forums for troubleshooting.

closed 2016-06-13 by RMerl

monosoul, 2016-08-12:
@RMerl So, today I've updated to 380.61 and seeing the same issue. **BUT**, now I was able to determine what causing problems with IPv6! Looks like it stops working when ssh is enabled for **WAN+LAN** .
For example, here's working configuration:
Enable ssh: **LAN only**
Allow SSH Port Forwarding: Yes
SSH service port: 443
Allow SSH password login: no
Enable SSH Brute Force Protection: yes
SSH Authentication key: <some key>

And here's not working configuration:
Enable ssh: **LAN + WAN**
Allow SSH Port Forwarding: Yes
SSH service port: 443
Allow SSH password login: no
Enable SSH Brute Force Protection: yes
SSH Authentication key: <some key>

If i'll turn on "WAN + LAN" for ssh after IPv6 connection is established - then it's working before next reboot. And it starts working again only after I'll enable ssh for LAN only again.

Could you, please, take a look at this issue?

monosoul, 2016-08-12:
Maybe it could be related to this commit: https://github.com/RMerl/asuswrt-merlin/commit/ec140f6452b5f96a181887a1b3281f6f74a2c673 ?