Asuswrt-Merlin/asuswrt-merlin.legacy #938: Asus RT-N6U, 380.59, Unable to edit description after using quote in it

yk11, 2016-06-13:
When putting in a quote into a description (not hostname), e.g. "Wife's laptop", that entry can no longer be changed.  This applies to both DHCP assignment view and client list view.  When clicking the picture icon, nothing happens.

Error is: "Uncaught SyntaxError: missing ) after argument list" in Advanced_DHCP_Content.asp:1

RMerl, 2016-06-13:
What browser are you using?

yk11, 2016-06-14:
It happens in both Firefox and Chrome.

I noticed that when I click on an entry's icon, the following error shows up in the console regardless of whether the popup works or not:
```
TypeError: evt.srcElement is undefined
show_app_table()
```

The syntax error I mentioned shows up before the above error.

yk11, 2016-06-14:
I was able to verify the issue.  When I manually escaped the single quote, the popup worked.
`popClientListEditTable('<mac address>', this, 'Sarah\'s Moto G', '192.168.0.25', 'DHCP')`

RMerl, 2016-06-14:
Confirmed.  The odd thing is this is the same syntax used throughout the rest of the firmware, but it only happens on the DHCP page.

RMerl, 2016-06-14:
Works properly on the networkmap's client list, but also broken on the WOL list.

yk11, 2016-06-15:
Although the code look very similar (potential for future refactoring :) ), the ACL and DHCP sections use `clientName`, while the WOL section does not.

I'm referring to Advanced_ACL_Content.asp:111-133, Main_WOL_Content.asp:145-167, and Advanced_DHCP_Content.asp:299-321.

commit 99d389a2 (2016-06-15): webui: Client names with quotes in them couldn't be edited in the networkmap popup (closes #938)
closed 2016-06-15 by RMerl