Asuswrt-Merlin/asuswrt-merlin.legacy #939: 380.59 on RT-AC87U HDD spin down problem

domnulnaspa, 2016-06-14:
I have a WD My Passport HDD attached to my AC87U and have had no issues until the last update. I even reset the spin down setting on the HDD and also tried to set a spin down time in Tools, on router. The disk keeps spinning and spinning.

domnulnaspa, 2016-06-15:
It seems to have start working again but now it has some issue with the media server. I have to change any setting in the media server section to force it to show up. Is this tied to the HDD spinning down?

xlstefan, 2016-06-15:
I also have the same problem on the RT-AC87U.. I have to turn off the media server and turn it on in order to see HDD on the network.

domnulnaspa, 2016-06-15:
I might be on to something with this issue, but I have to wait a while to see if it sticks. Being out of other ideas I set a static ip on my TV and it suddenly showed up. I have to check this for consistency to be sure it was the TV's fault, but you can give it a try.

domnulnaspa, 2016-06-20:
Nope, it didn't solve, furthermore the issue with the spin down is back. Something keeps the Hdd spinning and something else keeps hiding the media server. And I'm sure it's something in 380.59, had no issues with previous firmware.

RMerl, 2016-06-21:
Please head to the support forums for troubleshooting.  There hasn't been any change to the hdd spindown daemon in over 3 years.

closed 2016-06-21 by RMerl