Asuswrt-Merlin/asuswrt-merlin.legacy #940: Samba share's auth fail

CyrusNG, 2016-06-15:
I have 3TB's Hard disk connected RT-AC68u router with usb3.0. And I want to use it to make samba share with NTFS format. But I found I can NOT pass auth to access samba share drive with my created account & admin account, but anonymous can. Are there any idea on this?

RMerl, 2016-06-15:
Try the support forums.

CyrusNG, 2016-06-16:
I aleady find the problem:
I create one samba user with special character, and samba store wrong encrypted password into /tmp/etc/samba/smbpasswd. So I login samba with linux or mac, always auth fail. My router admin account with special charactors also have the same issue.

But I don't know whether it is samba or asuswrt or merlin issue, I hope Merlin team would take a look.

Anyway, I have one temp solution for someone face the same issue:
when you input the password at merlin web interface, you can try to input "\" first, before you input password. This works for me.
e.g.
my password : $1234abcd!
input password: \$1234abcd!

Hope this can help someone with the same problem.

vdanbv, 2016-06-20:
Erase

RMerl, 2016-07-28:
The webui is generally fragile to the use of non alpha-numeric characters.  This is a general design flaw, not something that is possible to fix without a major architectural change to the webui code - something outside the scope of this project.

closed 2016-07-28 by RMerl

rip1980, 2016-07-29:
Your SAMBA version is old and relies on deprecated features, forcing modern SMB clients to degrade security to make it work.  This has nothing to do with the gui/passwords, the SAMBA server should be brought up to date.

Details:  https://ubuntuforums.org/showthread.php?t=2323358