Asuswrt-Merlin/asuswrt-merlin.legacy #941: spontaneous reboot on RT-AC87U v.380.59

tmaone, 2016-06-18:
I have been observing my AC87 lately and I see that it restarts arbitrarily every, between a few (3-4) days to a week, without having the restart schedule set, and without any apparent error or other log in the syslog. Is this normal or to be expected? Is there any way to enable extra logging perhaps to investigate it a little further? I've done several (not many) modifications on top of asuswrt-merlin, such as ipset tables using a couple kmods from the ones provided. I also maintain optware-ng on a usb flash but do not init any of the installed packages, instead I handle everything with jffs scripts and post-configs. That would be: customising dnsmasq a bit (to support host-based ad-blocking), ddns script, post-mount for the optware which adds swap/jffs scripts backup/ad-block init and kills u2ec, lpd and lld2d, and adds dnscrypt-client wrapper. I do not seem to find any other log to indicate a fault in the above services.

One another end, I'd like to request if it is possible to maintain a version.conf-like file similar to the one in `release/src-rt/version.conf`, but for the currently released version. I made a jffs/cron script to check for a firmware update which (conveniently) fetches `https://raw.githubusercontent.com/RMerl/asuswrt-merlin/master/release/src-rt/version.conf` as it can be sourced directly into env-vars, but after a release the version is bumped into the next-release version. It would be really convenient to have such a file and in that case, I can perhaps attempt to modify and offer a PR for a patch to integrate such functionality to upstream. Would be glad to know if something like this is of interest. 

Many thanks for the effort and excellent work btw.

RMerl, 2016-06-18:
Please use the forums for troubleshooting purposes.

Having a version manifest isn't as simple as having just a file with a version number, as the latest version number isn't always the same for every models.  For instance, 380.60 will probably not be available for the RT-N66U nor the RT-AC66U as Asus's GPL releases are missing required components for these two models.  A fully fledged manifest would be required, showing different versions for different model.  At this point, such a thing isn't a priority (and would most likely need to be hosted on a VPS - hosting such polled files on Github is poor netiquette, as they're designed for hosting project data, not frequently polled manifest data.

closed 2016-06-18 by RMerl

tmaone, 2016-06-18:
Thanks for the reply! Didn't really know there exists a support forum. Will ask there too. It would be good to know very briefly though, a) if an internal reboot is to be anticipated and b) if possible, how to enable extra logging.

I see the reasons that make the version file complex enough to be considered. Thanks for explaining!

RMerl, 2016-06-18:
The router will reboot whenever the kernel or the init process crashes.  The only way to debug these is through a serial cable - a crashed kernel is unable to write to syslog.

tmaone, 2016-06-19:
Thanks again.  That explains very well why it restarts and also why Im not seeing anything. Cheers.