Asuswrt-Merlin/asuswrt-merlin.legacy #944: Cannot flash firmware on RT-N66U

jpuderer, 2016-06-22:
Tried flashing the latest build (RT-N66U_380.59_0) 

Router is running stock firmware: 3.0.0.4.380_3264-g79f6bde

I tried rebooting the router, resetting to factory settings, nothing works.

Is there a way to figure out why it's failing?

mstombs, 2016-06-22:
Read the pinned post on support forums
http://www.snbforums.com/threads/asus-firmware-will-change-the-verification-method.32357/
You can still use CFE based recovery methods, the Asus repair tool or CFE web console for this router apparently.

jpuderer, 2016-06-22:
Worked great using the "Asus firmware restoration tool".  Guess someone should update the flashing instructions on the Wiki?

mstombs, 2016-06-22:
Well if you know where in the wiki...  Current issue should be transient, but we can't have a beta 380.60 due to Asus not releasing the relevant files for Merlin to be able to build for mips routers, see [https://github.com/RMerl/asuswrt-merlin/issues/923](url)

jpuderer, 2016-06-23:
Yes, it's this page:
https://github.com/RMerl/asuswrt-merlin/wiki/Installation

RMerl, 2016-06-26:
There's no mention of it on the Wiki yet because the final details have yet to be finalized by Asus.

closed 2016-06-30 by RMerl