Asuswrt-Merlin/asuswrt-merlin.legacy #945: L2TP VPN Server support

hungyc, 2016-06-23:
Start from iOS 10, Apple has cancelled PPTP connectivity, is it possible to support L2TP VPN Server?

RMerl, 2016-06-23:
No plan to.  OpenVPN is the recommended VPN technology for Asuswrt.

davericher, 2016-06-24:
If thats the case this will be my last iphone

closed 2016-06-26 by RMerl

iMocW, 2016-06-30:
OpenVPN Client didn't update for a long time.Even not support iOS 8.

unsynaps, 2016-06-30:
OpenVPN client works fine in iOS. Using it now in 9.3.2 (current as of this post).

iMocW, 2016-06-30:
Official Client..??I tried yesterday and it just FC right after open it.

unsynaps, 2016-06-30:
https://itunes.apple.com/us/app/openvpn-connect/id590379981?mt=8

Some people are ranting and raving saying it does not work but it works without issue for me on the latest Merlin firmware using the config files it spits out.

averkiev, 2016-08-24:
I'm using the official OpenVPN client on iOS10. It also worked just fine on iOS9. Still I would rather not to rely on a third-party solution for my VPN needs and would much prefer native support of L2TP in router firmware.