Asuswrt-Merlin/asuswrt-merlin.legacy #949: Improve README.TXT

smitgd, 2016-06-26:
README.TXT leaves out several details for building rt-ac56u and related targets when building on Fedora 23 and earlier fedora versions.
Add package "gpref" to additional required fedora packages.  This avoids configuration error when building "minidlna".
No need to make symlinks in /opt but must set PATH correctly (project bin last):
export PATH=$PATH:[root-of-asuswrt-merlin]/release/src-rt-6.x.4708/toolchains/hndtools-arm-linux-2.6.36-uclibc-4.5.3/bin/
Manual steps involving autotools seem to be needed one time to do a successful build from a clean "git checkout master":
cd [root-of-asuswrt-merlin]/release/src-rt-6.x.4708/router/libdaemon
libtoolize
aclocal
autoheader
automake --force-missing --add-missing
autoconf
cd ../libxml2
mv configure.in configure.ac
vi configure.ac  (edit and remove or comment out AM_C_PROTOTYPES, i.e., 
dnl AM_C_PROTOTYPES)
libtoolize
aclocal
autoheader
automake --force-missing --add-missing
autoconf
cd ../..  (back to [root-of-asuswrt-merlin]/release/src-rt-6.x.4708
(The above operations on libdaemon and libxml2 only needed one time unless a clean checkout done again that wipes out the changes.)
make cleankernel clean [rt-ac56u |  rt-ac68u | rt-n18uhp]
Subsequent rebuilds for minor changes can skip the cleaning targets:
make [rt-ac56u |  rt-ac68u | rt-n18uhp]

Note: The above causes the fedora autotool/libtools to be used instead of the auto/libtools included in the project. If project bin path is moved first in the PATH this does not eliminate build errors, but causes project auto/libtools to be used. Neither does adding symlinks at /projects and /opt, as indicated by other build errors when project auto/libtools used. Even with these changes the build still fails (at least for me) unless the above manual fixes in libdaemon and libxml2 are done with the project bin set last in the PATH. Maybe I am missing something?

RMerl, 2016-06-26:
This file is actually from Asus' GPL.  I don't use Fedora myself, so that file isn't really maintained by me.

I probably should remove the file itself, as the "official" build documentation can be found on the Wiki.

smitgd, 2016-06-27:
Yes, the README.TXT leaves out setting the path/symlink for ARM toolchain (the old  version from years ago did). So removing it seems OK.
The wiki instructions for ubuntu look like they would mostly work for fedora and include the manual fixes for libdaemon and libxml2 and mention setting the ARM path.
However, the instruction for mint and debian don't explicitly mention the ARM path that I could see.
I would suggest referencing the wiki for build instructions in README-merlin.txt when README.TXT is deleted.

RMerl, 2016-07-20:
Rewrote README.TXT (commit db2eb76feec8302b6e52cc0381cd9edd171cab85) to be more up-to-date.  This was based on my own notes as I recently built a new development VM based on Ubuntu 16.04 LTS.

closed 2016-07-20 by RMerl