Asuswrt-Merlin/asuswrt-merlin.legacy #950: WAN DNS 1 cannot be set from UI

dannyboyw, 2016-06-27:
WAN DNS 1 could be set/was displayed correctly via web UI using release 380.58.  Release 380.59 appears not to be functioning correctly, as setting **Advanced Settings** under **WAN** section **WAN DNS Setting** entry **DNS Server1** and clicking **Apply** does not work.  Populating value for NVRAM name *wan_dns1_x* from CLI works!  
Ex: 
`nvram set wan_dns1_x=208.67.222.222`

Asuswrt-Merlin is excellent! Thank you for your efforts!

dmolavi, 2016-06-30:
It works for me on .59 on an AC66U, upgraded from .58. What HW are you running this on?

dannyboyw, 2016-06-30:
AC3200.  It works!  I am not certain what I was experiencing, but after upgrade DNS Server1 was unset and it definitely was set prior when on .58.  I tried to apply and it still came back blank/unset.  I set via CLI and refreshed page and it was set.  Now it seems to consistently be working.  Sorry for the trouble!

dannyboyw, 2016-07-06:
This is happening again!  The value is now blank!  I know I did `nvram commit` after I set it from CLI.  I also tested by setting diff values in UI and applying, which (I imagine) does a commit.  I see that NVRAM is getting dangerously low/out (i.e. UI says: *NVRAM usage 	65548 / 65536 bytes* and `nvram show` says *size: 65507 bytes (29 left)*) and I saw it go into the negative (e.g. when doing `nvram show` *...(-11 left)...*) at least once.  Is this causing the flakiness?  What is eating up all the NVRAM?  I should have plenty on an AC3200, right?  Should I submit another issue to help me troubleshoot maxed NVRAM?  I am not sending logs, etc. to NVRAM that I know of.  What are some other things to look for?

RMerl, 2016-07-08:
You are running out of free nvram.  This isn't a firmware bug, but a config issue on your router.

Most common source are old openvpn key/certs that haven't been migrated to JFFS because you haven't used the service since that migration implementation, or having a lot of network clients filling up the networkmap (once again, this has been migrated to jffs a few months ago).

Simplest method is to do a factory default reset, and manually reconfigure everything.  The other method is to export a text version of your nvram content, and manually look for any unused nvram setting, and clearing it.

The RT-AC3200 uses a lot of nvram due to the fact that it has three radios to configure.

Closing this, as this isn't caused by a firmware issue.

closed 2016-07-08 by RMerl

dannyboyw, 2016-07-10:
I really appreciate the detail on your response!  I went w/ the second option, "...manually look for unused nvram settings...", but took an automated approach.  Do you see anything wrong with doing something like the below?  I recovered almost 10MBs of NVRAM.  I just want to make sure the overall approach is correct, detect empty value and unset them.  More importatantly, do you deem this as safe?
```
...
# Iterate over name=value pairs unsetting empty values
#
# Counters for summary stats...
# Total pairs count
pairsCnt=0
# unset pairs count
unsetPairsCnt=0
#
for nmValPair in $(nvram show | sort)
do
  # increment pairs counter
  pairsCnt=$((pairsCnt+1))
  
  #
  # XXX: echo is for debugging, therefore commented out for normal operations
  #
  #echo "${logTag}: ($pairsCnt)  Pair: $nmValPair"
  #
  # Parse into name and value vars for reuse
  nm="$(echo "$nmValPair" | cut -d '=' -f 1)"
  vl="$(echo "$nmValPair" | cut -d '=' -f 2-)"
  #
  #nm="$( cut -d '=' -f 1 <<< "$nmValPair" )"
  #vl="$( cut -d '=' -f 2- <<< "$nmValPair" )"
  #
  
  #echo "${logTag}: ($pairsCnt)  Name: $nm"
  #echo "${logTag}: ($pairsCnt) Value: $vl"
  
  # value is blank/empty
  if [ -z  "$vl" ]; then
    # increment unset counter
    unsetPairsCnt=$((unsetPairsCnt+1))
    
    # unset/unload pair from NVRAM
    nvram unset $nm
    
    # Log pair instead of just the name to assist with debug/troubleshooting
    logMsg "($unsetPairsCnt/$pairsCnt) Unset unused pair: $nmValPair" ${logTag} ${logFilePath}
  fi

done
...
```

If you think this is appropriate, what about running it on a schedule, like weekly or daily?  The only problem I found thus far is that *cut* (probably something I am doing wrong) is chopping up my SSH keys into multiple entries/tests, but this does not result in valid settings getting *unset* because they have value/are not empty.  I understand IFS might be better suited to do this, but it has caveats as well.  If you see this as viable, I may tinker w/ it further when I get a chance.  

Log output sample:

```
nvram-conserv-a-tron: started on: Fri Jul  8 12:14:44 DST 2016
nvram-conserv-a-tron: Unsetting unused/empty pairs...
...
nvram-conserv-a-tron: (25/373) Unset unused pair: apps_dev=
nvram-conserv-a-tron: (26/374) Unset unused pair: apps_download_file=
nvram-conserv-a-tron: (27/375) Unset unused pair: apps_download_percent=
nvram-conserv-a-tron: (45/460) Unset unused pair: cloud_sync=
nvram-conserv-a-tron: (46/465) Unset unused pair: cstats_exclude=
nvram-conserv-a-tron: (47/466) Unset unused pair: cstats_include=
nvram-conserv-a-tron: (48/487) Unset unused pair: custom_usericon=
nvram-conserv-a-tron: (49/488) Unset unused pair: custom_usericon_del=
nvram-conserv-a-tron: (50/495) Unset unused pair: ddns_cache=: 
...
nvram-conserv-a-tron: (605/3524) Unset unused pair: wrs_app_rulelist=
nvram-conserv-a-tron: (606/3529) Unset unused pair: wrs_rulelist=
nvram-conserv-a-tron: Unset 606 of 3534 total pairs on: Fri Jul  8 12:16:54 DST 2016.  Logged details to: unset-empty-nvram-names.sh.log
```

**Thanks again for you time, efforts and patience!**