Asuswrt-Merlin/asuswrt-merlin.legacy #951: After last firmware update(on official ASUS firmware to, may be asus bug) incorrectly working SIP

PRIHLOP, 2016-06-27:
After last firmware update(on official ASUS firmware to, may be asus bug) incorrectly working RTSP Passthrough and may be SIP Passthrough.

RMerl, 2016-06-27:
Can you be more specific?  Asuswrt and Asuswrt-Merlin handle the passthrough settings differently.

PRIHLOP, 2016-06-27:
This bug true on RT-AC68U with Asuswrt-Merlin, and also true on RT-AC66U with Asuswrt. May be new binaries made this bug?

RMerl, 2016-06-27:
Myself and quite a few other users are using ATAs without any problem with 380.59.  I use a Cisco SP112 myself.

I recommend heading to the forums for troubleshooting.  While Asus recently changed things so that disabling passthrough will also block port 5060, that is not the case with my firmware.

PRIHLOP, 2016-06-28:
But you may fix it?
Bug is no dial tone for outgoing calls.

Thank you for your patience and development of this firmware.

RMerl, 2016-06-28:
If it works for everyone else, then the issue more likely to be with your configuration than with the firmware.

PRIHLOP, 2016-06-28:
May be you're right. If the problem is not repeat at others users, then you are right.

closed 2016-06-28 by PRIHLOP