Asuswrt-Merlin/asuswrt-merlin.legacy #953: Real Time Traffic Stats, click anywhere on graph?

dmolavi, 2016-06-30:
See the attached screenshot...

Basically on the real-time graph, you can click anywhere, and it'll show you the value where you clicked vs the value of the upload/download at that particular point in time. I believe the latter to be the more useful value for users.  If you need anything tested, I'd be willing to lend an assist.
[image]

willyb321, 2016-07-28:
I think this'd be great. 👍

RMerl, 2016-08-24:
Unlikely to ever be implemented, as it would require studying the whole graphing code, something I've never touched before.  After a quick glance, it looks non-trivial to implement.