Asuswrt-Merlin/asuswrt-merlin.legacy #957: Support ASUS RT-AC88U Firmware version 3.0.0.4.380.3341

henryj, 2016-07-01:
So from the official description:

> ASUS has been dedicated to cooperate with third party developers to come up with more innovative features. 
> To comply with regulatory amendments, we have modified firmware verification rule to ensure better firmware quality. 
> This version is not compatible with all previously released ASUS firmware and uncertified third party firmware.

Wondering if we're gonna support the new rule soon in the coming release?

RMerl, 2016-07-01:
http://www.snbforums.com/threads/important-asuswrt-merlin-380-59-and-older-not-compatible-with-asus-3-0-0-4-380_3000-or-newer.32962/

henryj, 2016-07-01:
Great. So the up coming 380.60 release will fix this. I guess we can keep this issue open till then. 

(Off topic, seems that forum is not accessible from Australia? I have to jump on my US VPN to see that thread...)

RMerl, 2016-07-01:
SNB experienced some stability issues these past few days, that might be why.  Otherwise, it could be that your IP is blacklisted by Cloudflare.

I know of at least one regular who's also from Australia and doesn't seem to have any issue.

RMerl, 2016-07-01:
Here's the official explanation of the recent SNB problems:

RMerl, 2016-07-01:
Here's the official explanation regarding the SNB problems of late:

http://www.snbforums.com/threads/is-something-wrong-with-the-site.33399/

Email the site owner with your info if you still have problems accessing it.

closed 2016-07-07 by RMerl