Asuswrt-Merlin/asuswrt-merlin.legacy #958: Roaming assistant threshold higher than -70dBm is ineffective

robert-alfaro, 2016-07-01:
The recent change in commit [4db5dd7](https://github.com/RMerl/asuswrt-merlin/commit/4db5dd7c0df3614ab244752ca34bb72dde068d30) for roaming assistant does allow settings the threshold value to a higher -50dBm, however, testing shows the disconnect only happens when passing the -70dBm bound (less than -70) instead of less than -50dBm. 

Perhaps there is a lower level threshold limiter?

RMerl, 2016-07-02:
When I asked users to test it, they said it worked, however I did have my doubts on whether the wireless driver allowed such values to work.  If it truly doesn't work, then it means it's an internal limitation, one which I cannot change.  It's a possibility, would explain why Asus chose such a conservative limitation.

RMerl, 2016-07-08:
Reverting patch since it doesn't work (commit 8cbb504886db686bc42f952904a7e76be9f5d911)

closed 2016-07-08 by RMerl