Asuswrt-Merlin/asuswrt-merlin.legacy #959: [380.59] [RT-AC66U] No clients in network map

waspinator, 2016-07-02:
I'm using Firmware 380.59 on a RT-AC66U and I can't seem to be able to see any clients in the `Network `Map section

I know I have at least 2 wifi devices hooked up and 3 wired, but nothing shows up in the list.

[image]

xlstefan, 2016-07-02:
Use refresh button and they will show up.

marodor, 2016-07-02:
check if you dont have any devices with a "quote" in his hame

waspinator, 2016-07-05:
@xlstefan I have tried pressing refresh, but it has not added any clients. It still shows `No data in table`

@marodor After opening my Firefox console, and clicking refresh I see the following messages:

    GET XHR http://10.10.10.1/status.asp?_=123461784532 [HTTP/1.0 200 Ok 12ms]
    GET XHR http://10.10.10.1/update_networkmapd.asp?_=123461784532 [HTTP/1.0 200 Ok 2ms]
    GET XHR http://10.10.10.1/ajax_status.xml?hash=0.523461784532[HTTP/1.0 200 Ok 23ms]
    GET XHR http://10.10.10.1/update_clients.asp?_=123461784532 [HTTP/1.0 200 Ok 25ms]
    GET XHR http://10.10.10.1/update_networkmapd.asp?_=123461784532 [HTTP/1.0 200 Ok 3ms]
    21:28:31.780 The character encoding of a framed document was not declared. The document may appear different if viewed without the document framing it.1 apply.cgi
    GET XHR http://10.10.10.1/status.asp?_=1467681929085 [HTTP/1.0 200 Ok 10ms]
    ...

quanganhdo, 2016-07-18:
Same problem here with 380.59 and RT-N66U. Before I upgraded to this version there had been no problems.

thezoggy, 2016-07-20:
380.59 on RT-AC66U -- client list is fine here. (both the right panel and the popup)

RMerl, 2016-07-28:
Open your web browser console and look for any Javascript error.

waspinator, 2016-07-28:
@RMerl I don't seem to have any javascript errors. The only output I see is some CSS issues and the XHR requests. Tried this on Firefox and Safari.

RMerl, 2016-07-28:
People are reporting similar issues with the stock firmware, and I'm unable to reproduce the issue here, so someone else (probably Asus) will have to fix this.

z3r0tnd, 2016-08-11:
I had the same Issue and could reproduce it. 
The network devices will not shown if you use the IP-address from the router as the web address.
Only if you use the DNS Name router.asus.com it will show up all of your network devices.

thezoggy, 2016-08-11:
380.59 on RT-AC66U -- client list is fine here. (both the right panel and the popup)
navigating to webui via ip address works fine here on firefox/win7... what os/browser?

waspinator, 2016-08-12:
I've tried in Firefox, Safari, and Edge. Bare browsers, no extensions.

Does Asus have a repo/issue tracker somewhere to report this?

RMerl, 2016-08-12:
Make sure you can reproduce the issue with an original Asus firmware, then try contacting them through networking_support <at> asus <dot> com.

The stock firmware also has a Feedback page on the webui to submit reports.

thezoggy, 2016-09-22:
380.61 -- I've now ran into this issue on Firefox, yesterday it was fine.. only thing that changed was me updating dhcp/wireless ip/hostnames... note that while firefox is having problems the page still works in chrome...

loading the homepage the console shows:
```
SyntaxError: JSON.parse: unexpected character at line 1 column 1 of the JSON data
var store = (win.name ? JSON.parse(win.name) : {});

client_function.js -- Line 78
```

the lan > dhcp server page throws the same, with also
```
TypeError: clientList is undefined
if(clientList[clientMac]) {

Advanced_DHCP_Content.asp -- Line 243
```

thezoggy, 2016-09-22:
used chrome to go back and rename some dhcp hostnames.. sorted page.. went back to firefox and via firebug had it clear related cookies for the site.. closed out of tab, open new one.. its working again.

whats odd is that when using the java console and telling it to toss window.name or window.top.name to the console log it just reports back 'z' while chrome does undefined.  now that its working again, the output is now an empty string.

RMerl, 2016-09-26:
Can anyone retest with 380.62?  I fixed a few issues in networkmap.  Doubt they were related, but still worth re-testing since this issue seem to be very random.

barney34, 2016-09-27:
Hey Merlin the issue is still happening in 380.62 I can confirm.  Three browsers, 3 different PC, and one VM where browsers had never been used before so not a cache issue.   Thank you!

RMerl, 2016-09-27:
Then we will have to wait for Asus to fix it.  I cannot reproduce the problem.

recrunchi, 2016-10-27:
I have the same problem. 380.62_1 it still happens. Tonite i was editing the names in "LAN / DHCP Server", and then when i want to see connected clients in Network Map/View List tab, it shows nothing. I re-edited it, and nothing. Then i tried what thezoggy said, (clearing cache with firebug and closing/opening new tab in firefox), and still nothing happened...

pianoquintet, 2016-11-04:
Same problem here with 380.62_1.  Also, all my manually-assigned ip addresses have disappeared from the LAN-DHCP Server page.

pianoquintet, 2016-11-04:
By way of an update, the problem seems fixed on my router after I switched off and then on the Apps Analysis under Adaptive QoS as well as the ipv6 connection under ipv6.  Not sure which of the two caused the interference but thought I would offer this for troubleshooting purposes.

recrunchi, 2016-11-04:
pianoquintet, what kind of update?

pianoquintet, 2016-11-04:
Hi, not sure what happened to my post. My update was that the issue seems
to be fixed on my router after I disabled and re-enabled the Apps Analysis
under Adaptive QoS as well as the ipv6 connection under ipv6. No idea which
of the two things bears any relationship to the client list issue but
thought I would offer my experience, if relevant.

RMerl, 2016-12-09:
Hopefully resolved with eced399d36c481ae55a9d504dc93854b5a78d225 as I managed to be lucky enough to reproduce the problem for a second time in all these months.

thezoggy, 2016-12-18:
using 380.63_2 the problem still happens, seeing console report back:
`SyntaxError: JSON.parse: unexpected character at line 1 column 1 of the JSON data`
`TypeError: clientList is undefined`

RMerl, 2016-12-19:
That bug was fixed in 380.64.

closed 2016-12-19 by RMerl

thezoggy, 2016-12-19:
doh, thanks. will update

waspinator, 2016-12-19:
I'm now seeing some clients, but not most. My subnet mask is 255.255.0.0, but I'm only seeing hosts on the subnet the router is on. Is this expected behavior? For example the router is on 10.10.0.X, but most clients are on 10.10.1.X.

RMerl, 2016-12-19:
That's a known limitation, the networkmap service is hardcoded for a /24.  Making it match the configured subnet would require some low-level resdesign, something I'd rather let Asus do.

hinder90, 2018-01-06:
I hate to resurrect this a year after this was closed, but I am actually having this same symptom but what I suspect is a very different cause. I have determined that if a host is plugged directly into a LAN port it shows up on the map and the QOS list. However, all wireless and other wired hosts connected by way of an uplinked switch are not visible. Like other users, this problem manifested itself without precedent. I have had the same firmware 380.69 for at least a month now and have been running your version of ASUSWRT for the past 2-3 years and I have never seen this problem. Not surprisingly, it is not limited to any browser or platform. In fact, using the ASUS iOS admin tool I see the same results. I'm not sure what to do about it.

hulmevv, 2018-01-10:
I too am now having this issue occur! On an RT-AC5300 running 380.69.  VERY FRUSTRATING!

Pairodocks, 2018-01-12:
I think this is related to this thread. I am new here and a total novice to settings on routers.  I have an RT-N66U.  

I just upgraded the firmware for the first time,  Before the upgrade I could see my client list.  Now all I see is the number of clients.  When I click "View List", nothing happens.  I have Refreshed the page.  I have clicked on "Refresh" on the ASUS router page.  I have tried changing browsers from Firefox to IE.  Nothing seems to work.  I can no longer see what is connected Wirelessly.  I have nothing connect via LAN.  Someone please Help. Thanks in advance.

Pairodocks, 2018-01-12:
Also - The firmware version is **3.0.0.4.380_8120**

IamSharpy, 2018-01-20:
Me too... 3.0.0.4.380_8120

wattersd, 2018-01-25:
I installed my first merlin firmware ever today (382.1_2) - hoping to fix another issue I was experiencing with the stock asuswrt (was unable to connect wirelessly after about 12 hours uptime -  clients with correct credentials were being rejected with incorrect pass key - a reboot fixed the issue but not a good long term fix). I've no noticed that I'm not getting anything in the client list table (I was straight after the merlin firmware was installed - but no longer. I've tried Safari and Chrome - no client list in either. Tried hitting <Refresh> - nothing. I think I do have apostrophes in some of my client names (no idea how to remove them now I can't see them).

Is this now tagged as an Asus bug @RMerl - I was able to see my client list prior to flashing the merlin firmware?

Happy to try to provide any further context / debugging if needed.

byterex, 2018-01-31:
I just received my new router and have the same problem here with RT-AC88U and 382.1_2. The client list is completely corrupted, it changes very often (empty, full, empty,...) even if the clients are connected and stable. Not all the clients are shown in the list and the LAN clients not at all.
The clients are also mixed up, name and MAC address, this is really terrible (name from 1 client and MAC from another). Really annoying! Too much money for such a crap....
asus shame on you!