Asuswrt-Merlin/asuswrt-merlin.legacy #960: OpenVPN Static key auth problem with DNS

iMocW, 2016-07-03:
There's no DNS conf writing to /tmp/resolv.conf (means didn't apply custom DNS define in .ovpn file(right?)) after connecting to a static key auth method and there's a configure line "dhcp-option dns 8.8.8.8" (WAN default DNS isn't 8.8.8.8 and I want to router to use 8.8.8.8 automaticly after connect to OpenVPN (PPTP just do this fine)(do not use a script,a script might can do this but It feels weird)) in .ovpn file .Windows and iOS is auto applied DNS setting correctly after connection processing.
Ask me if you need my .ovpn and server.conf file content,and more information.
PS.I know there is DNS setting if using TLS.And I know it works.I have to use static key to go through firewall.

iMocW, 2016-07-03:
BTW , Is there anybody know if Android does not support static key auth via official OpenVPN client ?

cricket007, 2016-07-13:
I am using static-key authorization on my router with "Respond to DNS" set to Yes and a custom configuration option to `push "dhcp-option DNS 8.8.8.8"`, and when I connect with Tunnelblick on OS X I see...

```
# /etc/resolv.conf
search openvpn
nameserver openvpn.server.ip.address
nameserver 8.8.8.8
```

Alternatively, if you set "Connect to DNS Server automatically" to No under the WAN page, you can set your DNS servers there, then if you direct DNS queries through the VPN, you will also have that  DNS resolver.

RMerl, 2016-09-03:
I can't find any interaction in the source code between the presence of a static key and how nameservers are handled.

iMocW, 2016-09-03:
I don't know. I have to use this script to solve my problem:
`
#!/bin/sh
socyz1=`echo $1 | grep n2`
socyz2=`echo $1 | grep p2`
socyz=server
if [ ""$socyz1"" == """" ]; then
if [ ""$socyz2"" == """" ]; then
socyz=client
fi
fi

if [ $socyz == client ]; then
  logger "openvpn-event:this is client event"
  logger \$script_type=$script_type
  if [ -e /tmp/etc/openvpn/client?/config.ovpn ]; then
    clientNO=`find /tmp/etc/openvpn -name client? | head -n 1 | sed -e "s/openvpn\//\n/" | tail -n 1`
    if [ $script_type == down ]; then
#      Change 198.126.159.13 to your VPS IP and uncomment it,or use edit-host.sh to add your VPS IP.
#      route del -net 198.126.159.13 netmask 255.255.255.255
    fi
        
    if [ $script_type == route-up ]; then
      /jffs/scripts/chnroutes.sh
      #This command is design for static key mode,DNS has to be added to .ovpn file using "dhcp-option DNS 8.8.8.8"(etc)
      #从.ovpn文件读取DNS并自动设置到dnsmasq服务。
      DNSfc2=`cat /tmp/etc/openvpn/$clientNO/config.ovpn | grep DNS | tail -n 1 | sed -e "s/DNS /\n/" | tail -n 1`
      DNSfc1=`cat /tmp/etc/openvpn/$clientNO/config.ovpn | grep DNS | tail -n 2 | head -n 1 | sed -e "s/DNS /\n/" | tail -n 1`
      
      if [ ""$DNSfc1"" != """" ]; then
        echo "nameserver $DNSfc1" >/tmp/resolv.conf
        if [ ""$DNSfc2"" != ""$DNSfc1"" ]; then
          if [ ""$DNSfc2"" != """" ]; then
            echo "nameserver $DNSfc2" >>/tmp/resolv.conf
          fi
        fi
        logger "dnsmasq server restarted with DNS in .ovpn file"
        service restart_dnsmasq
      else
        logger "No DNS in .ovpn file"
      fi
    fi
  else
    logger "can not load .ovpn file, exit now"
  fi
else
  logger "openvpn-event:server event,exit now"
fi

logger exit:$script_type
`
This scripit works fine.

iMocW, 2016-09-03:
Format seems wrong. BTW, Where are you from ? CA ?