Asuswrt-Merlin/asuswrt-merlin.legacy #961: Firmware upgrade unsuccessful

voltechs, 2016-07-03:
I'm trying to flash my router with the latest version. Coming from the most recent version.

I get `Firmware upgrade unsuccessful. This may result from incorrect image or error transmission. Please check the version of firmware and try again.` after trying to upload the `trx` file.

Product ID: RT-AC68U
Signature Version: 1.112
Firmware Version: 3.0.0.4.380_3264-g79f6bde

Trying to use version `RT-AC68U_380.59_0`

What's up?

eyre0950, 2016-07-04:
You can downgrade by using Firmware Recovery mode。
The new version is not compatible with all previously released ASUS firmware and uncertified third party firmware.

marodor, 2016-07-05:
You can downgrade by using Firmware Recovery mode and doing this https://www.asus.com/support/faq/1000814/

mane, 2016-07-07:
Do you know which firmware I should downgrade to? I've got an RT-AC87U.

Thank you.

RMerl, 2016-07-07:
Please use the support forum at SNBForums for support discussions.

This is not a bug in the firmware.

closed 2016-07-07 by RMerl