Asuswrt-Merlin/asuswrt-merlin.legacy #962: pre-mount does not hold boot sequence

k-emil, 2016-07-04:
pre-mount running lengthy e2fsck does not prevent the system from starting.
Log:
Aug  1 02:00:06 syslogd started: BusyBox v1.20.2
Aug  1 02:00:06 kernel: klogd started: BusyBox v1.20.2 (2016-05-10 11:44:43 EDT)
Aug  1 02:00:06 kernel: start_kernel
Aug  1 02:00:06 kernel: Linux version 2.6.22.19 (root@asus) (gcc version 4.2.3) #1 Tue May 10 11:45:15 EDT 2016
Aug  1 02:00:06 kernel: CPU revision is: 00019749
Aug  1 02:00:06 kernel: Determined physical RAM map:
Aug  1 02:00:06 kernel:  memory: 07fff000 @ 00000000 (usable)
Aug  1 02:00:06 kernel:  memory: 08000000 @ 87fff000 (usable)
Aug  1 02:00:06 kernel: Built 1 zonelists.  Total pages: 585216
Aug  1 02:00:06 kernel: Kernel command line: root=/dev/mtdblock2 noinitrd console=ttyS0,115200
Aug  1 02:00:06 kernel: Primary instruction cache 32kB, physically tagged, 4-way, linesize 32 bytes.
Aug  1 02:00:06 kernel: Primary data cache 32kB, 4-way, linesize 32 bytes.
Aug  1 02:00:06 kernel: Synthesized TLB refill handler (20 instructions).
Aug  1 02:00:06 kernel: Synthesized TLB load handler fastpath (32 instructions).
Aug  1 02:00:06 kernel: Synthesized TLB store handler fastpath (32 instructions).
Aug  1 02:00:06 kernel: Synthesized TLB modify handler fastpath (31 instructions).
Aug  1 02:00:06 kernel: PID hash table entries: 2048 (order: 11, 8192 bytes)
Aug  1 02:00:06 kernel: CPU: BCM5300 rev 1 pkg 0 at 600 MHz
Aug  1 02:00:06 kernel: Using 300.000 MHz high precision timer.
Aug  1 02:00:06 kernel: console [ttyS0] enabled
Aug  1 02:00:06 kernel: Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Aug  1 02:00:06 kernel: Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
Aug  1 02:00:06 kernel: Memory: 238608k/131068k available (2610k kernel code, 22664k reserved, 517k data, 196k init, 131072k highmem)
Aug  1 02:00:06 kernel: PCI: Enabling device 0000:01:00.1 (0004 -> 0006)
Aug  1 02:00:06 kernel: PCI: Fixing up bus 1
Aug  1 02:00:06 kernel: PCI/PCIe coreunit 1 is set to bus 2.
Aug  1 02:00:06 kernel: PCI: Fixing up bridge
Aug  1 02:00:06 kernel: PCI: Fixing up bridge
Aug  1 02:00:06 kernel: PCI: Enabling device 0000:02:00.1 (0004 -> 0006)
Aug  1 02:00:06 kernel: PCI: Fixing up bus 2
Aug  1 02:00:06 kernel: NET: Registered protocol family 2
Aug  1 02:00:06 kernel: Time: MIPS clocksource has been installed.
Aug  1 02:00:06 kernel: IP route cache hash table entries: 4096 (order: 2, 16384 bytes)
Aug  1 02:00:06 kernel: TCP established hash table entries: 16384 (order: 5, 131072 bytes)
Aug  1 02:00:06 kernel: TCP bind hash table entries: 16384 (order: 4, 65536 bytes)
Aug  1 02:00:06 kernel: TCP: Hash tables configured (established 16384 bind 16384)
Aug  1 02:00:06 kernel: TCP reno registered
Aug  1 02:00:06 kernel: highmem bounce pool size: 64 pages
Aug  1 02:00:06 kernel: squashfs: version 3.2-r2 (2007/01/15) Phillip Lougher
Aug  1 02:00:06 kernel: io scheduler noop registered (default)
Aug  1 02:00:06 kernel: HDLC line discipline: version $Revision: 4.8 $, maxframe=4096
Aug  1 02:00:06 kernel: N_HDLC line discipline registered.
Aug  1 02:00:06 kernel: Serial: 8250/16550 driver $Revision: 1.90 $ 4 ports, IRQ sharing disabled
Aug  1 02:00:06 kernel: serial8250: ttyS0 at MMIO 0xb8000300 (irq = 8) is a 16550A
Aug  1 02:00:06 kernel: serial8250: ttyS1 at MMIO 0xb8000400 (irq = 8) is a 16550A
Aug  1 02:00:06 kernel: PPP generic driver version 2.4.2
Aug  1 02:00:06 kernel: MPPE/MPPC encryption/compression module registered
Aug  1 02:00:06 kernel: NET: Registered protocol family 24
Aug  1 02:00:06 kernel: PPPoL2TP kernel driver, V0.18.3
Aug  1 02:00:06 kernel: PPTP driver version 0.8.5
Aug  1 02:00:06 kernel: Physically mapped flash: Found 1 x16 devices at 0x0 in 16-bit bank
Aug  1 02:00:06 kernel:  Amd/Fujitsu Extended Query Table at 0x0040
Aug  1 02:00:06 kernel: Enabling Spansion 65nm mode, writesize = 512 bytes
Aug  1 02:00:06 kernel: Physically mapped flash: CFI does not contain boot bank location. Assuming top.
Aug  1 02:00:06 kernel: number of CFI chips: 1
Aug  1 02:00:06 kernel: cfi_cmdset_0002: Disabling erase-suspend-program due to code brokenness.
Aug  1 02:00:06 kernel: Flash device: 0x2000000 at 0x1c000000
Aug  1 02:00:06 kernel: Creating 5 MTD partitions on "Physically mapped flash":
Aug  1 02:00:06 kernel: 0x00000000-0x00040000 : "pmon"
Aug  1 02:00:06 kernel: 0x00040000-0x01fe0000 : "linux"
Aug  1 02:00:06 kernel: 0x00176990-0x017a0000 : "rootfs"
Aug  1 02:00:06 kernel: 0x01fe0000-0x02000000 : "nvram"
Aug  1 02:00:06 kernel: 0x017e0000-0x01fe0000 : "jffs2"
Aug  1 02:00:06 kernel: Found an  serial flash with 0 0KB blocks; total size 0MB
Aug  1 02:00:06 kernel: sflash: found no supported devices
Aug  1 02:00:06 kernel: dev_nvram_init: _nvram_init
Aug  1 02:00:06 kernel: sdhci: Secure Digital Host Controller Interface driver
Aug  1 02:00:06 kernel: sdhci: Copyright(c) Pierre Ossman
[... 170 lines cut ...]
Jul  4 10:11:57 dnsmasq-dhcp[245]: DHCPREQUEST(br0) 192.168.200.217 00:8c:fa:85:c8:e1 
Jul  4 10:11:57 dnsmasq-dhcp[245]: DHCPACK(br0) 192.168.200.217 00:8c:fa:85:c8:e1 Toshiba
Jul  4 10:12:24 dnsmasq-dhcp[245]: DHCPREQUEST(br0) 192.168.200.199 88:9f:fa:33:cd:3d 
Jul  4 10:12:24 dnsmasq-dhcp[245]: DHCPACK(br0) 192.168.200.199 88:9f:fa:33:cd:3d halszka
Jul  4 10:13:34 dnsmasq-dhcp[245]: DHCPREQUEST(br0) 192.168.200.253 00:1f:c6:62:1a:4d 
Jul  4 10:13:34 dnsmasq-dhcp[245]: DHCPACK(br0) 192.168.200.253 00:1f:c6:62:1a:4d dev20
Jul  4 10:14:27 dnsmasq-dhcp[245]: DHCPINFORM(br0) 192.168.200.199 88:9f:fa:33:cd:3d 
Jul  4 10:14:27 dnsmasq-dhcp[245]: DHCPACK(br0) 192.168.200.199 88:9f:fa:33:cd:3d halszka
Jul  4 10:16:17 dnsmasq-dhcp[245]: DHCPREQUEST(br0) 192.168.200.203 5c:0a:5b:24:89:f3 
Jul  4 10:16:17 dnsmasq-dhcp[245]: DHCPACK(br0) 192.168.200.203 5c:0a:5b:24:89:f3 android-86a37bfc7d1eb476
Jul  4 10:16:26 dnsmasq-dhcp[245]: DHCPREQUEST(br0) 192.168.200.217 00:8c:fa:85:c8:e1 
Jul  4 10:16:26 dnsmasq-dhcp[245]: DHCPACK(br0) 192.168.200.217 00:8c:fa:85:c8:e1 Toshiba
Jul  4 10:17:24 dnsmasq-dhcp[245]: DHCPREQUEST(br0) 192.168.200.199 88:9f:fa:33:cd:3d 
Jul  4 10:17:24 dnsmasq-dhcp[245]: DHCPACK(br0) 192.168.200.199 88:9f:fa:33:cd:3d halszka
Jul  4 10:18:34 dnsmasq-dhcp[245]: DHCPREQUEST(br0) 192.168.200.253 00:1f:c6:62:1a:4d 
Jul  4 10:18:34 dnsmasq-dhcp[245]: DHCPACK(br0) 192.168.200.253 00:1f:c6:62:1a:4d dev20
Jul  4 10:20:55 dnsmasq-dhcp[245]: DHCPREQUEST(br0) 192.168.200.217 00:8c:fa:85:c8:e1 
Jul  4 10:20:55 dnsmasq-dhcp[245]: DHCPACK(br0) 192.168.200.217 00:8c:fa:85:c8:e1 Toshiba
Jul  4 10:21:05 dnsmasq-dhcp[245]: DHCPREQUEST(br0) 192.168.200.203 5c:0a:5b:24:89:f3 
Jul  4 10:21:05 dnsmasq-dhcp[245]: DHCPACK(br0) 192.168.200.203 5c:0a:5b:24:89:f3 android-86a37bfc7d1eb476
Jul  4 10:21:19 kernel: kjournald starting.  Commit interval 5 seconds
Jul  4 10:21:19 kernel: EXT3 FS on sda3, internal journal
Jul  4 10:21:19 kernel: EXT3-fs: mounted filesystem with ordered data mode.
Jul  4 10:21:19 syslog: USB ext3 fs at /dev/sda3 mounted on /tmp/mnt/sda3
Jul  4 10:21:19 usb: USB ext3 fs at /dev/sda3 mounted on /tmp/mnt/sda3.

RMerl, 2016-07-08:
It should not prevent the rest of the firmware from running, only the disk from mounting until the script has completed.  This is working as intended, I see a 10 minutes gap between the moment the pre-mount script gets launched and the moment the disk gets mounted under /tmp/mnt/sda3.

closed 2016-07-08 by RMerl

k-emil, 2016-07-08:
Right, but please take into account, that overlay is on the disk, so the router boots up with the configuration being far from required.

From: Eric Sauvageau 
Sent: Friday, July 08, 2016 7:23 AM
To: RMerl/asuswrt-merlin 
Cc: k-emil ; Author 
Subject: Re: [RMerl/asuswrt-merlin] pre-mount does not hold boot sequence (#962)

It should not prevent the rest of the firmware from running, only the disk from mounting until the script has completed. This is working as intended, I see a 10 minutes gap between the moment the pre-mount script gets launched and the moment the disk gets mounted under /tmp/mnt/sda3.

RMerl, 2016-07-08:
Router events are run async, so there's no way for the script to prevent the rest of the firmware from running.  Also, pre-mount isn't run by the boot process specifically, but at any point a disk gets plugged in, and udev triggers the event.  It's not a linear process.

k-emil, 2016-07-12:
Right. Implementing overlay I will have to keep the “default” config making the router deaf and silent, opening required config after mounting the disk in question.
That is apparently the only way to go.
Thanks and apologies for raising spurious “bug” 

From: Eric Sauvageau 
Sent: Friday, July 08, 2016 3:35 PM
To: RMerl/asuswrt-merlin 
Cc: k-emil ; Author 
Subject: Re: [RMerl/asuswrt-merlin] pre-mount does not hold boot sequence (#962)

Router events are run async, so there's no way for the script to prevent the rest of the firmware from running. Also, pre-mount isn't run by the boot process specifically, but at any point a disk gets plugged in, and udev triggers the event. It's not a linear process.