Asuswrt-Merlin/asuswrt-merlin.legacy #963: Problem with scheduled reboots

smallfishy, 2016-07-04:
Just to report I'm experiencing failed router start ups about 25-33% of the time from scheduled reboots on the RT-AC68U. Most of the time the router is operational afterwards but then sometimes for no apparent reason I will struggle to connect to the router (no WiFi).

flitzjoy, 2016-07-05:
Having this same problem, same model and similar rate of success.

danhrainey, 2016-07-08:
Me Three! RT-AC68U running 380.58 scheduled reboot hangs and wifi does come back.

compmas2, 2016-07-17:
I can also confirm this issue.  It also occurs on the regular ASUS firmware for the last couple version too.

RMerl, 2016-07-31:
Can you retest this with one of the 380.61 test builds?  I wasn't able to reproduce it on my own RT-AC68U.

flitzjoy, 2016-07-31:
I schedule my router to reboot on Saturdays. And for two weeks in a row, using 61 version it worked fine. Maybe it's fixed on latest asus GPL.

RMerl, 2016-07-31:
I suspected this might be the case, as 3831 added a few missing libraries to build_temp_rootfs(), which is the temporary filesystem used during a shutdown.

RMerl, 2016-08-03:
Can another of the other reporters confirm this is indeed fixed with 380.61 beta?

Lord-Phoenix, 2016-08-03:
I'm have same issue on AC68U with latest original firmware from ASUS. Never had it before.

RMerl, 2016-08-15:
Haven't received any other feedback, so since it was resolved for those who reported on it, I'll consider this to be fixed as of 380.61.

closed 2016-08-15 by RMerl

danhrainey, 2016-08-20:
Hi RMerl,
Have been traveling this month. I will give it a try adn report when i return and i will report back

apotts, 2016-10-03:
I have had this problem for the last couple of major releases, and it continues with 380.61

RT-AC68U too.

NoSync, 2016-10-06:
Still having this problem on both 380.61 and 380.62.

Phobooky, 2017-01-07:
Having the same problem with RT-AC68U 380.64_1.

daha76, 2017-01-08:
Having the same problem with RT-N66U 380.64_2.

jjames55, 2017-08-23:
This does not seems to be fixed at all, I have two RT-AC68Us and one RT-N66U all experiencing frequent hangs when restarting from a scheduled reboot on 380.66..

smallfishy, 2017-08-23:
Can I suggest raising this issue with Asus if the Asus firmware is showing the same issue.
The reporting function in most of the latest Asus firmware seems to be effective. I no longer have the RT-AC68U but the DSL-AC88U, which has had some minor bugs. Asus were very responsive to my reports - producing beta firmware within a few weeks - but then that is one of their flagship products at the moment. I think a lot of their routers share very similar firmware though.

jjames55, 2017-08-25:
Dunno with the asus, running Merlins from scratch. Funny thing, i have 3 routers they seem to experience the hangs at the same time, and not every day. For now i have had to remove the scheduled nightly reboot. 
Also when the sympton is across two router versions (AC68/N66) i guess the common denominator could be the firmware. I have upped the logging level to see if it writes more...
the last entry always seems to be 
"ntp: start NTP update"

talespin416, 2017-10-08:
I am using AC66U and i have this problem ever since i upgraded to 380.68 directly from a version that was maybe 3 versions ago. So far 100% of my weeklt scheduled reboots have failed.
One time everything was seemingly fine but one client had no internet. After mesaing with things for over an hour, i decided to reboot the router and voila. 
I love the scheduled reboot feature because it used to allowed running the router for months at a time without so much as even logging into the router interface at all.
Any idea when this might be fixed?

ro0ter, 2017-12-11:
I have same problem on AC-68u, latest stock ASUS firmware. I was planning to migrate to RMerl agian (RT-N16 had RMerl, no longer supported, time to get a new router) but unfortunately I see it has same problem.

Anyone knows why is this caused? I have rich media saved on my usb3.0 hdd, a laptop for uploading pictures to the respective hdd and a kodi for consumnig videos/photos... need to reboot it every now and then but the scheduled reboot feature seems to render the router unable to connect to PPPoE sometimes (currently 10~15% of scheduled reboots). Hope to get a chance and investigate what`s wrong this time.

omegadcuk1965, 2017-12-17:
Hi been on an older version of Merlin for ages and just upgraded to 380.69 on RT-AC68U.  Same problem ( I assume).  On a reboot router appears to be down, but on closer inspection WIFI is down, WAN/LAN is up and running.  So it would appear that the WIFI is actually dead on reboot?

I didn't need to reboot again to fix it just connected in to the front end via LAN from laptop directly to port on router.

- Logged in
- Reapplied password to WIFI's 
- Reduced logging to Error only (noticed this was different from before)
- Turned off HTTPS login (I had just enabled this so it was suspect)

It APPEARS to be rebooting and schedule restarting ok, 3 out of 3 so far.  So fingers crossed ;)  

I will post back if it starts happening again.
Hope this helps?