Asuswrt-Merlin/asuswrt-merlin.legacy #964: OpenVPN Client can't auto apply ifconfig configuration.

iMocW, 2016-07-05:
There is the line "ifconfig 12.12.12.2 255.255.255.0" in the .ovpn file.
When uploaded this file to router,this configuration shows in custom configuration and do not auto apply to "Local/remote endpoint addresses" or "Tunnel address/netmask". 

And When connected to server.The "config.ovpn" file in /tmp/etc/openvpn/client1/ shows two "ifconfig" line under both "# Automatically generated configuration" and "# Custom Configuration".

Althrough the conf will be overwritten by Custom Configuration(higher priority from mytest),I still hope you can fix this bug(maybe a bug)

Situation above:tested in TUN/Static key(Local/remote endpoint addresses) and TAP/Static key(Tunnel address/netmask)

And another bug:Althrough "redirect-gateway def1" shows under Automatically generated configuration,but It isn't work.
Have to set(RedirectInternet traffic/All traffic/Gateway:12.12.12.1) manually to access internet via VPN.

iMocW, 2016-07-07:
The second problem may not a bug. Windows client have to set this manually too.

RMerl, 2016-07-08:
If you need to replace an existing config line, you should use a postconf script:

https://github.com/RMerl/asuswrt-merlin/wiki/Custom-config-files

The custom field is meant to add to, not to replace or modify the existing config content.  This is the expected behaviour and not a bug.

iMocW, 2016-07-08:
not custom-config-file..have you seen my problem? I said config file mean custom configuration auto apply to /tmp/etc/openvpn/client*/client.ovpn. You may need to re-read my problem.

RMerl, 2016-07-08:
You specifically said that you were using the Custom Configuration section:

```
Althrough the conf will be overwritten by Custom Configuration
```
The content of that section doesn't overwrite anything, it gets appended to the config file generated by the router.  You need to use a postconf config script to manipulate the content of the generated config file.

RMerl, 2016-07-08:
When you import an existing .ovpn file, the content is parsed by the router, and recognized keywords are used to configure the associated nvram settings.  Any unrecognized keyword gets added to the Custom Config section.  Once the import is done, the ovpn file is deleted.

After that, when the router generates a client config, it creates it from scratch based on the content of the nvram settings, and appends at the end the content of the Custom Configuration section.

ifconfig is not a recognized keyword, that's why it ends up in the Custom section, and it might conflict with any ifconfig entry that gets generated by the router.

closed 2016-07-31 by RMerl

misse, 2017-05-05:
How does the config.ovpn get generated? Because mine generates the following:
# Custom Configuration
client
dh [[INLINE]] -----BEGIN DH PARAMETERS-----
(DH DATA removed)
-----END DH PARAMETERS-----
and the client can't start, citing " Options error: Unrecognized option or missing parameter(s) in config.ovpn:24" and line 24 is the first line of DH data.

iMocW, 2017-05-06:
I don't know how .ovpn file has generated. I know there's a problem that hasn't be resolved for a long time. You just configured it incorrectly. DH is a part of TLS handshaking, ask your server provider to provide the correct .ovpn file. 如果你来自中国，则不建议使用TLS模式与国外OpenVPN服务器进行连接，而应该使用Static Key. 
The problem is "dhcp-option DNS" configuration still need to apply "manually" using an auto script.