Asuswrt-Merlin/asuswrt-merlin.legacy #967: Firmware is not compatible and uncertified

didok3003, 2016-07-08:
Hello, following the readme.txt instructions, I uploaded the file to flash the firmware, howere an error was shown:
[image]

This is the file I used:
[README-merlin.txt](https://github.com/RMerl/asuswrt-merlin/files/354747/README-merlin.txt)

> Simply flash it like any regular update.  You should not need to 
reset to factory defaults (see note below for exceptions).
You can revert back to an original Asus firmware at any time just
by flashing a firmware downloaded from Asus's website.

any idea?

RMerl, 2016-07-08:
Asus recently changed the validation due to new regulation requirements from bodies such as the FCC.

See this announcement for more information:

http://www.snbforums.com/threads/important-asuswrt-merlin-380-59-and-older-not-compatible-with-asus-3-0-0-4-380_3000-or-newer.32962/

didok3003, 2016-07-09:
There is a beta but for RT-AC68U, is it working for mine too? May I wait a while?

RMerl, 2016-07-09:
Yes, the beta would work, or you can flash 380.59 using Asus's Firmware Recovery tool.

didok3003, 2016-07-09:
I have installed 380.59 using Asus's Firmware Recovery tool. It worked perfectly.

Is there any subscription or notification that will be sent once 380.60 is ready?

Thanks!

RMerl, 2016-07-10:
No.  I always announce new releases to Twitter, SNBForums and on the website.

didok3003, 2016-07-13:
Everything fine now, thanks for the support.

closed 2016-07-13 by didok3003