Asuswrt-Merlin/asuswrt-merlin.legacy #968: dnsmasq cause DHCP not worked.

zw963, 2016-07-09:
Merlin version. 380.59

Deivce: MIPS AC66U

When I start dnsmasq with following config:

1. echo conf-dir=/opt/etc/dnsmasq-conf > /jffs/configs/dnsmasq.conf.add

2. /opt/etc/dnsmasq-conf 目录下文件
admin@RT-AC66U-20F0:/tmp/mnt/sda/entware/etc# tree -sh dnsmasq-conf/
dnsmasq-conf/
|-- [735K] accelerated-domains.china.conf
|-- [2.2K] bogus-nxdomain.china.conf
|-- [ 25] foreign_domains.conf
`-- [ 39K] google.china.conf
Those config is basically from  [dnsmasq-china-list](https://github.com/felixonmars/dnsmasq-china-list) project.  except foreign_domains.conf with following content.

```sh
server=/#/127.0.0.1#1082  # this is ss-tunnel listening port.
```

3. reboot route.

4. DHCP not worked. could not connect to route if not settings a static ip address.

Thanks.

RMerl, 2016-07-09:
This is a configuration issue, not a bug in the firmware.  I recommend asking on the support forums at SNBForums.

closed 2016-07-09 by RMerl

zw963, 2016-07-13:
several days,. i create a post there, but no one answer me question,could you please give me a clue about how to discover this? thanks