Asuswrt-Merlin/asuswrt-merlin.legacy #969: Parental Restriction - Time-based usage

un33k, 2016-07-09:
Right now, the specific time must be set in the AiProtection Time Scheduling.

It would be great to have a feature where each device could be configured up-to max number of hours per day.

This way, kids can have 2 hours of internet access per day at any time. That combined with the schedule-based is a convenient way to limit internet access to kids.

RMerl, 2016-07-20:
This would require extensive changes to the existing implementation, which would then make it very difficult to keep that portion of the code in sync with Asus.   This is outside the scope of this project, sorry.

closed 2016-07-20 by RMerl

pepeEL, 2016-07-20:
un33k maybe write to ASUS support and write on SNB Forum. There are people from ASUS.

Starmann, 2016-07-21:
This would be hard to do anyways. You would need to track time spent using the internet. First of all, it would be unable to how long a user was on a page. Secondly, it would be unable to detect if someone was actively using the computer (Maybe windows was updating.). If a computer downloads a webpage, how much time should be used? Might I suggest instead you figure out about how much data your kids use in 2 hours and limit their data usage to that. This isn't a perfect system. I suggest instead you try and monitor _What_ your kids are doing. Computer usage can be very educational. I spent hours and hours reading Wikipedia as a kid and I turned out fine (Says the person who is talking about parenting philosophy in the issues section of a router firmware.). 4 hours spent reading about the Sumerians is certainly better than 2 hours spent on Facebook. I do not wish to criticize your parenting so I apologize if it came off that way.