Asuswrt-Merlin/asuswrt-merlin.legacy #971: DDNS namecheap issues

dhighet, 2016-07-10:
I've been on chat with NameCheap support a few times, so I know my settings have to be correct. Though when trying to submit the DDNS settings I receive "Request Error! Please try again." Here is the log file

`Jul 10 14:00:27 watchdog: start ddns.
Jul 10 14:00:27 rc_service: watchdog 260:notify_rc start_ddns
Jul 10 14:00:27 ddns update: ez-ipupdate: starting...
Jul 10 14:00:27 ddns update: connected to dynamicdns.park-your-domain.com (104.219.249.25) on port 80.
Jul 10 14:00:27 ddns update: unknown return code: 400
Jul 10 14:00:27 ddns update: server response: Bad Request
Jul 10 14:00:27 ddns update: asusddns_update: 1`

rlj4214, 2016-07-10:
Me to.
I gave up

RMerl, 2016-07-13:
It's possible that Namecheap might have changed the API.  In that case, someone else would have to look into it. I always refused to add additional DDNS services for this very reason - they are very time-consuming to support.  Namecheap was only added because someone else had taken the time to implement it.  If it's broken and no one else can fix it, then I will have to remove it.  Namecheap users will have to either use a custom DDNS script, or switch to DNS-o-Matic (which, I assume, does support Namecheap).

The quite antique DDNS client used by Asuswrt kinda limits us.  For instance, that client has no support for SSL/TLS.

dhighet, 2016-07-13:
Actually, it appears they do. I've set up the router page per [these guidelines](https://www.dnsomatic.com/wiki/dd-wrt) and was prompted that the registration was successful. Though, on the dnsomoatic  services page, it is returning an error for namecheap.

> SETDNSHOSTeng1No Records updated. A record not Found;1380091No updates; A record not Found;true

jaredfolkins, 2016-07-14:
Namecheap still works for me. Here is an example wget request based on the script I have. I actually have several segments to my bash script for different dynamic domain names through name cheap.

# Request
```bash
wget --no-check-certificate -qO - "https://dynamicdns.park-your-domain.com/update?host=@&domain=example.com&password=332c5f08a28c32c5f08a28c2c5f08a28c&ip=" 
```

# Reponse
```bash
<?xml version="1.0"?><interface-response><Command>SETDNSHOST</Command><Language>eng</Language><IP>your.ip.address.here</IP><ErrCount>0</ErrCount><ResponseCount>0</ResponseCount><Done>true</Done><debug><![CDATA[]]></debug></interface-response>
```

# NameCheap DDNS Dashboard
[image]

jtp10181, 2016-07-14:
Namecheap support is working fine for me. I have had it setup for a while.
I just forced a refresh and it was successful. Using 380.59 on an AC56

dhighet, 2016-07-21:
It looks like it was my method of updating multiple records at once. The reps instructed me to use **@ -a www -a *** which is returning the error. If I have just, say, **@** it will update without issue. What is the correct format for multiple records?

RMerl, 2016-07-22:
Use a custom DDNS script to update additional records.

Closing this, as this is not an issue with the firmware code.

closed 2016-07-22 by RMerl

gbailey4, 2017-06-20:
@dhighet did you put together a script that solves this? I'm encountering the same issue.