Asuswrt-Merlin/asuswrt-merlin.legacy #972: Motd Custom Config Files (motd and motd.add)

pmagid, 2016-07-13:
I know its doable to change the motd now by overwriting /tmp/etc/motd.  I thought it might be cleaner to add motd and motd.add custom config file to the list of understood custom configs that can be changed by adding a file to /jffs/configs/.

RMerl, 2016-07-20:
motd is usually useful with a multi-user system.  What kind of usage would require a single user router to use a customized motd?

pmagid, 2016-07-20:
motd is also useful to set legal warnings (as appropriate for the laws in ones local jurisdiction) for crackers etc.   that is just one example.  its use is not only to send messages to authorized users of a multi-user system, although that is is motd's intended and original purpose.

RMerl, 2016-07-21:
It's not as simple as with other config files, because in this case the existing file is in read-only flash, while other config files are dynamically generated in RAM.

It's doable, but will require more work, and is really low priority at this point.