Asuswrt-Merlin/asuswrt-merlin.legacy #974: Error occurred when using the golang net library

xinghusp, 2016-07-13:
I tried to run a network program on the router,but it return error.
the same program is running successfully on the arm router (ac56u) which also run the merlin 
on the n66u with merlin,it can't create the listen port.but on the same router with openwrt,it running success.

i use the golang to build the program.
[image]

xinghusp, 2016-07-13:
the merlin version is 380.59_release

RMerl, 2016-07-13:
This isn't a problem with the firmware itself.  I recommend asking on the support forums at SNBForums, someone might be able to help you with it.  Most likely, you will need to tweak your cross compile makefile to work with the older kernel / uClibc used by the MIPS platform.

closed 2016-07-13 by RMerl