Asuswrt-Merlin/asuswrt-merlin.legacy #980: 5-1 Ghz bug

frostalf, 2016-07-23:
Have an Asus RT-AC3200, if you leave the 5-1 ghz controll channel set to auto or 36. On reboots it uses channel 6 as the control channel. However setting it to 48 makes it not change to channel 6. This bug is present in the Vanilla firmware from asus version 3.0.0.4.380.3479. I didn't experienced this problem in the vanilla firmware version 3.0.0.4.378.9529 or at least didn't notice it, as well exists in the merlin asus firmware version 380.59. I would report it to Asus but they lack a reporting place for bugs so I am putting it here, so that if others have come across the same problem they know its not just them and what I did to fix it temporarily. I have only tested it a few times, once in the vanilla firmware, once in merlin, and 2 times when I set it to 48. I haven't tested if it does it on the other channels.

RMerl, 2016-07-28:
This doesn't make sense, since channel 6 is a 2.4 GHz channel, not a 5 Ghz channel.  There's no way the 5 GHz radio could be using channel 6.

frostalf, 2016-08-01:
While I understand this, it reports that is what channel its trying to use on my screen and when it does it, it causes the 5-1 Ghz to not be available at all. Also, when its doing this it doesn't appear to interfere with the 2.4Ghz band as the devices that use that seem to work just fine. I have never seen this before which is why I am submitting it lol

RMerl, 2016-08-01:
So most likely it doesn't use channel 6, it's simply displaying the wrong channel.

On which page does it show that channel?

RMerl, 2016-08-03:
Can't reproduce it on my RT-AC3200.  Tested both Asus's 3.0.0.4.380_3479 and Asuswrt-Merlin 380.61 beta, tried both setting it to "36" and to "auto", and after a reboot I always end up getting channel 36 shown both on the Wireless config page as well as the Wireless Log.

Try doing a factory default reset and reconfiguring manually - your nvram content might be corrupted.  The RT-AC3200 is a bit short on nvram, so it can easily fill up, leading to truncated settings.

frostalf, 2016-08-24:
sorry for the long delay in response, I have done as you said and it appears to have fixed the problem. Thanks for the help, and the page it showed the channel on was where I would go and edit the settings for wifi. It showed in the drop down box that it was set to channel 6, and as I said before I know this shouldn't be possible. Anyways thanks again.

closed 2016-08-24 by frostalf