Asuswrt-Merlin/asuswrt-merlin.legacy #982: kernel supports transparent proxy?

zw963, 2016-07-26:
Hi, when I run `ss-redir` which installed from entware-ng with `-u` arg, 
get following error message

```sh
admin@RT-AC66U-20F0:/tmp/mnt/sda/entware/etc# ss-redir -c ./ss-redir1.json -u
 2016-07-05 18:12:09 INFO: initialize ciphers... aes-256-cfb
 2016-07-05 18:12:09 INFO: UDP relay enabled
 2016-07-05 18:12:09 ERROR: [udp] setsockopt IP_TRANSPARENT
```

I create issue to this project, the maintainer told me, 

``Make sure your kernel supports transparent proxy`

I hope to know, current kernel transparent proxy is support ?

Thanks.

RMerl, 2016-07-28:
What specific kernel setting would that be?

blackfuel, 2016-08-08:
Does this document say what's going on?
https://www.kernel.org/doc/Documentation/networking/tproxy.txt

I think here is the kernel support:
CONFIG_NETFILTER_XT_MATCH_SOCKET=m
CONFIG_NETFILTER_XT_TARGET_TPROXY=m

zw963, 2016-08-09:
Sorry for that i don't know what kernel support this too.

all I know is the maintainer told me, kernel is not support. : )

closed 2016-11-06 by RMerl