Asuswrt-Merlin/asuswrt-merlin.legacy #983: xt_DSCP.c fails for RT-N66U compilation

eponymous4473, 2016-07-26:
I am using the latest git clone and while building for RT-N66U I received the following: 

...
  CC [M]  net/netfilter/xt_HL.o
  CC [M]  fs/reiserfs/tail_conversion.o
net/netfilter/xt_DSCP.c: In function 'target':
net/netfilter/xt_DSCP.c:38: error: dereferencing pointer to incomplete type
net/netfilter/xt_DSCP.c:43: error: dereferencing pointer to incomplete type
net/netfilter/xt_DSCP.c: In function 'target6':
net/netfilter/xt_DSCP.c:59: error: dereferencing pointer to incomplete type
net/netfilter/xt_DSCP.c:64: error: dereferencing pointer to incomplete type
net/netfilter/xt_DSCP.c: In function 'checkentry':
net/netfilter/xt_DSCP.c:75: error: dereferencing pointer to incomplete type
net/netfilter/xt_DSCP.c: At top level:
net/netfilter/xt_DSCP.c:90: error: invalid application of 'sizeof' to incomplete type 'struct xt_DSCP_info' 
net/netfilter/xt_DSCP.c:99: error: invalid application of 'sizeof' to incomplete type 'struct xt_DSCP_info' 
make[6]: *** [net/netfilter/xt_DSCP.o] Error 1
make[6]: *** Waiting for unfinished jobs....
...

eponymous4473, 2016-07-26:
I just cloned tag 380.59 and got the same type of errors on xt_DSCP.o.  I imagine I must be doing something wrong...?

RMerl, 2016-07-28:
Works for me.  Head to the forums for support, something in your build process is incorrect.

closed 2016-07-28 by RMerl