Asuswrt-Merlin/asuswrt-merlin.legacy #984: OpenVPN password gets corrupted after update

haraldinho2000, 2016-07-27:
I use OpenVPN on my RT-AC68U. I noticed that there is a bug in the router when you update a VPN users password: it corrupts the already established passwords.

Reproduction scenario:
1) configure OpenVPN
2) add a user with a password
3) click apply
4) confirm that the VPN connectivity works

Next, add another VPN user
5) add a new user with a different password
6) click apply
7) try to use the VPN connectivity with the first created user and notice that it fails to authenticate

As long as you create multiple users in one batch and then click apply it is fine. As soon as you add a user, the whole set of previously created user/password combinations get corrupted.

I use Merlin 380.59.

RMerl, 2016-07-28:
Works for me.

Check the content of the vpn_serverx_clientlist nvram, is it displaying correctly?

Also make sure not to use any unusual characters such as % or $.  Asuswrt's webui is quite fragile to these non alpha-numeric characters.

RMerl, 2016-08-24:
Closing since there's been no further feedback, and the issue couldn't be reproduced.

closed 2016-08-24 by RMerl

RMerl, 2016-08-24:
Might possibly be fixed by commit 0d51130a01bf58ecb9773cc6d47a48bb4d99ac3d if the issue only occurred if any username or password was over 32 characters long.

haraldinho2000, 2016-08-24:
> Might possibly be fixed by commit 0d51130 if the issue only occurred if any username or password was over 32 characters long.

Hi Eric,

Yes that was indeed the case. I use VPN passwords that are longer than 32 chars. Thanks for fixing it and for finding the root cause even after you could not reproduce. 

Sorry for my late response but I am on a long vacation. 

Kind regards from Monument Valley, USA

Harald Diesveld

Verstuurd vanaf mijn iPhone