Asuswrt-Merlin/asuswrt-merlin.legacy #985: Asus loose internett

spot88, 2016-07-28:
Running the newest version on RT-AC3200 , i loose internett maybe 2 times a day. Just reboot Asus won't work, i need to reboot my Fiber Optic modem and then Asus, after that everything works until next time.

spot88, 2016-07-28:
Anyone else been having the same problem, or might have a idea of what the problem is?

RMerl, 2016-07-28:
Please head to the support forums for troubleshooting.  Most of the time these are ISP-related, not firmware-related, especially if they occur multiple times per day.

closed 2016-07-28 by RMerl

spot88, 2016-07-28:
@RMerl Thanks for your answer, i work at my ISP, and the problem is not there. When i loose internett i get internett from the originally provided box. 
- I'm now using original Asus firmware and so far it's working. 

I was on the latest Merlin Firmware before.