Asuswrt-Merlin/asuswrt-merlin.legacy #987: Question re: update on FCC compliance & firmware updates

ekilah, 2016-07-29:
The internet is sparse with information on this, so I thought I'd just ask - what's the status of this situation with ASUS, the FCC, and your software? 

I'm looking to buy an ac88u, and I have seen mumblings that I'd need to install Merlin prior to doing anything else with the unit, including hooking up the WLAN to the internet, because the new firmware it'll download as an update blocks third party updates for now because of some FCC regulations.

Can you give me/us an update, since I can't tell how old that information is? I'd really appreciate it!

RMerl, 2016-07-30:
Asus's official statement: http://www.snbforums.com/threads/asus-firmware-will-change-the-verification-method.32357/

Note regarding Asuswrt-Merlin: http://www.snbforums.com/threads/important-asuswrt-merlin-380-59-and-older-not-compatible-with-asus-3-0-0-4-380_3000-or-newer.32962/

ekilah, 2016-07-30:
Great, thank you for the forum links, I read through a lot of them to catch up.

To summarize for anyone getting here from Google and wondering the same:

The 380.60 Merlin betas currently available (july 2016) can be flashed on top of the new Asus firmware. The 380.59 Merlin stable release can only be applied to the new Asus firmware if you use some recovery tool to get back to an older Asus firmware first, and that workaround's functionality might go away in the future.

ekilah, 2016-07-30:
and thank you @RMerl for the work! I'm new to the community but am excited to try out your mod.

closed 2016-07-30 by ekilah