Asuswrt-Merlin/asuswrt-merlin.legacy #995: Add a Dockerfile that provides an easy to use build environment

ograff, 2016-08-10:
Reduce the barrier to entry of entry for contribution / development by providing a Dockerfile that will build a docker image containing all necessary dependencies to build asuswrt-merlin. 

The docker image's entry point should be tools/build_all.

Update the development documentation to give an example using docker to build a firmware image.

ograff, 2016-08-10:
@RMerl I'd be happy to work on this and submit a PR if its something you think you would like having in the repo.

RMerl, 2016-08-12:
Asuswrt's build environment is quite complex and fragile.  I don't want to touch that build system, don't feel like spending time implementing any alternative build system either, and having to deal with fixing it every time a change upstream breaks it.

closed 2016-08-12 by RMerl