Asuswrt-Merlin/asuswrt-merlin.legacy #997: QUIC issue

haoel, 2016-08-11:
I currently use **ASUS RT-AC68U 1900M** router, and upgraded the `RT-AC68U_380.61_0.trx` firmware.

I active the VPN(PPTP) connection in router, and enable some static route rules and the `dnsmasq.conf.add` for different DSN servers.

Everything looks fine except to access the Google website by Chrome (Safari and Firefox no problem)

After I did some investigation, I found the Chrome uses [QUIC](https://en.wikipedia.org/wiki/QUIC) protocol to access the www.google.com or www.youtube.com website, and this cause the problem.

I made the following tests:

1) Using my laptop connect to the VPN directly, and Chrome can load the Google and Youtube website successfully via QUIC protocol. So, I think both my laptop and VPN haven't the problem.  

2) I disabled the Chrome's QUIC protocol - `chrome://flags/#enable-quic` , and Chrome using HTTPS, everything's fine. （scenario: laptop -> router -> vpn -> Google）

So, I think the problem might be the router has some problem with QUIC protocol, and some articles said the router's QoS setting could make the QUIC problem, but I've never enabled the QoS setting in router.

Anyone can help on this?

RMerl, 2016-08-11:
Please use the forums for support requests.

closed 2016-08-11 by RMerl

pjhartin, 2016-09-06:
I know this issue is closed, but I wanted to add that this is something I observed back in the .59 on my ASUS 68U, and am still observing. It makes using Chrome unbearable at times, especially on mobile. As stated, disabling QUIC locally via ://flags does provide an improvement, indicating that something to do with the QUIC protocol is affecting Merlin firmwares.