Asuswrt-Merlin/asuswrt-merlin.legacy #999: 380.61, SSH enabled on LAN+WAN breaks IPv6

monosoul, 2016-08-13:
Hi!

This is related to issue https://github.com/RMerl/asuswrt-merlin/issues/936 .
If i'll turn on "WAN + LAN" for ssh after IPv6 connection is established - then it's working till next reboot. And it starts working again only after I'll enable ssh for "LAN only" again.

For example, here's working configuration:
Enable ssh: LAN only
Allow SSH Port Forwarding: Yes
SSH service port: 443
Allow SSH password login: no
Enable SSH Brute Force Protection: yes
SSH Authentication key:

And here's not working configuration:
Enable ssh: LAN + WAN
Allow SSH Port Forwarding: Yes
SSH service port: 443
Allow SSH password login: no
Enable SSH Brute Force Protection: yes
SSH Authentication key: 

This problem first appeared in **380.59** firmware, so I guess it could be somehow related to this commit https://github.com/RMerl/asuswrt-merlin/commit/ec140f6452b5f96a181887a1b3281f6f74a2c673 .

monosoul, 2016-08-13:
Temporal fix for issue:
Enable port forwarding for ssh port to router's IP and iptable rule to accept connections from wan.

RMerl, 2016-08-13:
I'll have to see if I can reproduce it with an HE tunnel, as this is the only way I can test anything related to IPv6.

monosoul, 2016-08-13:
@RMerl OK, let me know if I can help you somehow.

kether667, 2016-08-15:
I have exactly same problem and can reliably reproduce. ipv6 works only with `enable ssh` set to LAN

RMerl, 2016-09-05:
What doesn't work specifically?

monosoul, 2016-09-05:
@RMerl router couldn't receive IPv6 parameters (prefix, dns etc.) from ISP.

RMerl, 2016-09-05:
I'm afraid I can't test this then, I only have access to an IPv6 tunnel, and it's working fine here.

monosoul, 2016-09-06:
@RMerl maybe I could help you somehow? Grab some logs or something?

RMerl, 2016-09-06:
See if there's anything in your system log indicating a failure.  But at a technical level, I cannot see what could lead to such a problem, unless something is really weird with the iptables rules.

kether667, 2016-09-06:
Before i found this ticket I've described my issue on the forum:
http://www.snbforums.com/threads/ipv6-issue.34077/#post-275513

In my logs http://pastebin.com/TALN4wy0 there is no indication of any problem.
i've been pointed to ipv6_ns_drop setting, but that did not have any effect.

As soon as i've set `enable ssh` to lan i got ipv6 ip address within 2-3 seconds and it works seamlessly ever since.

monosoul, 2016-09-06:
@RMerl i guess it's indeed should be related somehow to ip6tables. I'll compare rules in case of ssh enabled on lan only and on lan+wan later today. Also, I'll compare logs for both cases.

monosoul, 2016-09-07:
So, there's definitely a BIG difference in ip6tables rules when ssh enabled for lan + wan.
Here's list of rules when SSH enabled for LAN only: http://pastebin.com/5zdHvZMd
And here's list of rules when SSH enabled on LAN + WAN: http://pastebin.com/5zdHvZMd

@RMerl also, when I switch SSH from "LAN + WAN" to "LAN only" those INPUT rules are being added instantly, even before router manages to receive IPv6 settings from ISP, so I guess those missing rules are a root of our problem.

RMerl, 2016-09-08:
I wonder if the issue isn't a syntax error happening only on older MIPS routers such as yours, as it uses an older iptables version.  Could explain why it didn't happen here with my RT-AC88U and a 6in4 tunnel.

Can you check the content of the filter*.txt files in /tmp?  One of them will contain the IPv6 rules that should get applied to the filter table.

monosoul, 2016-09-10:
@RMerl yep, the one that doesn't get applied is `filter_rules_ipv6`. Error is:

    ip6tables-restore ./filter_rules_ipv6
    ip6tables-restore v1.3.8: Couldn't load match `recent':File not found

    Error occurred at line: 14`

Here is what it looks like: http://pastebin.com/9n6A0YfY

RMerl, 2016-09-10:
So the real cause is SSH Brute Force Protection, and not WAN access itself.  The MIPS version of iptables has separate versions of the "recent" netfilter module for IPv4 and IPv6, and the IPv6 version isn't currently enabled.

monosoul, 2016-09-10:
@RMerl will you enable it in the next version?

RMerl, 2016-09-10:
Yes, as long it works (currently compiling a test build, I was already doing tests on it).  Otherwise, I will simply remove the BFP rules from ip6tables, and limit it to IPv4.

monosoul, 2016-09-10:
@RMerl nice, thank you. Hope it'll work! :)

RMerl, 2016-09-10:
In the mean time as a work around, simply disable SSHD Brute Force Protection.

commit 72c8e1c8 (2016-09-10): iptables: Add missing ip6t_recent module for IPv6 (MIPS only) (fixes #999)
closed 2016-09-10 by RMerl