Asuswrt-Merlin/asuswrt-merlin.legacy #1010: Web listening on port 80

pculka, 2016-08-24:
Hi, I've upgraded to latest FW on AC66-u and I cannot force the frontend to stop listening on port 80, where I want to bind nginx as a reverse proxy to.

> admin@Bumbo:/tmp/home/root# lsof -i :80
COMMAND   PID  USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
httpd     253 admin    4u  IPv4   1669      0t0  TCP router.asus.com:www (LISTEN)
httpd     253 admin    5u  IPv4   1670      0t0  TCP localhost.localdomain:www (LISTEN)

also netstat -p doesn't work for some reason and I had to install lsof ...

/tmp/lighthttpd.conf does not even list listening on port 80, only server.port=8082

my chain input of IPTABLES contains:
> ACCEPT     tcp  --  anywhere             router.asus.com     ctstate DNAT tcp dpt:www

/jffs/scripts/firewall-start is currently empty

I'm a bit desperate and don't know where to continue next :(

RMerl, 2016-08-24:
Set it to HTTPS only, under System -> Administration.

pculka, 2016-08-24:
Thanks Merlin! sometimes the most simple solutions work the best, heh

closed 2016-08-24 by pculka