Asuswrt-Merlin/asuswrt-merlin.legacy #1011: tcp: make challenge acks less predictable

Thomas1Tseng, 2016-08-25:
This vulnerability allows off-path attackers to infer if any two hosts are
communicating using a TCP connection and, if so, further infer TCP sequence
numbers. An attacker can leverage this to trigger connection terminations
or data injection.

kernel\linux-3.12.14\src\net\ipv4\tcp_input.c
 /* rfc5961 challenge ack rate limiting */
-int sysctl_tcp_challenge_ack_limit = 100;
+int sysctl_tcp_challenge_ack_limit = 1000;
 
 int sysctl_tcp_stdurg __read_mostly;
 int sysctl_tcp_rfc1337 __read_mostly;
@@ -3244,13 +3244,18 @@  static void tcp_send_challenge_ack(struct sock *sk)
 	/* unprotected vars, we dont care of overwrites */
 	static u32 challenge_timestamp;
 	static unsigned int challenge_count;
-	u32 now = jiffies / HZ;
+	u32 count, now = jiffies / HZ;
 
 	if (now != challenge_timestamp) {
+		u32 half = (sysctl_tcp_challenge_ack_limit + 1) >> 1;
+              challenge_timestamp = now;
-		challenge_count = 0;
+		ACCESS_ONCE(challenge_count) = half +
+				prandom_u32_max(sysctl_tcp_challenge_ack_limit);
 	}
-	if (++challenge_count <= sysctl_tcp_challenge_ack_limit) {
+	count = ACCESS_ONCE(challenge_count);
+	if (count > 0) {
+		ACCESS_ONCE(challenge_count) = count - 1;
 		NET_INC_STATS_BH(sock_net(sk), LINUX_MIB_TCPCHALLENGEACK);
 		tcp_send_ack(sk);
 	}

kernel\linux-3.12.14\src\include\linux\random.h
+　  /**
+    * prandom_u32_max - returns a pseudo-random number in interval [0, ep_ro)
+    * @ep_ro: right open interval endpoint
+    *
+    * Returns a pseudo-random number that is in interval [0, ep_ro). Note
+    * that the result depends on PRNG being well distributed in [0, ~0U]
+    * u32 space. Here we use maximally equidistributed combined Tausworthe
+    * generator, that is, prandom_u32(). This is useful when requesting a
+    * random index of an array containing ep_ro elements, for example.
+    *
+    * Returns: pseudo-random number in interval [0, ep_ro)
+    */
+    static inline u32 prandom_u32_max(u32 ep_ro)
+    {
+    	return (u32)(((u64) prandom_u32() * ep_ro) >> 32);
+    }

Thomas1Tseng, 2016-08-25:
finish job

closed 2016-08-25 by Thomas1Tseng

RMerl, 2016-08-25:
The issue only affects kernels starting with release 3.6 - RFC5961 wasn't present in 2.6.22 or 2.6.36.