Asuswrt-Merlin/asuswrt-merlin.legacy #1014: Problem with compile RT-AC88U-please help

pepeEL, 2016-08-27:
Hi
I done all as is describe in wiki and i have a problem with building image RT-AC88U

   kernel   
(echo '.NOTPARALLEL:' ; cat /home/piotrek/asuswrt-merlin/release/src-rt-7.14.114.x/src/linux/linux-2.6.36/Makefile) |\
	make -C /home/piotrek/asuswrt-merlin/release/src-rt-7.14.114.x/src/linux/linux-2.6.36 -f - SRCBASE=/home/piotrek/asuswrt-merlin/release/src-rt-7.14.114.x/src BASEDIR=/home/piotrek/asuswrt-merlin/release/src-rt-7.14.114.x ARCH=arm zImage
make[4]: Wejście do katalogu '/home/piotrek/asuswrt-merlin/release/src-rt-7.14.114.x/src/linux/linux-2.6.36'
scripts/kconfig/conf --silentoldconfig arch/arm/Kconfig
  CHK     include/linux/version.h
  CHK     include/generated/utsrelease.h
  UPD     include/generated/utsrelease.h
  Generating include/generated/mach-types.h
awk: symbol lookup error: awk: undefined symbol: mpfr_z_sub
/home/piotrek/asuswrt-merlin/release/src-rt-7.14.114.x/src/linux/linux-2.6.36/arch/arm/tools/Makefile:8: polecenia dla obiektu 'include/generated/mach-types.h' nie powiodły się
make[5]: *** [include/generated/mach-types.h] Błąd 1
/home/piotrek/asuswrt-merlin/release/src-rt-7.14.114.x/src/linux/linux-2.6.36/arch/arm/Makefile:285: polecenia dla obiektu 'archprepare' nie powiodły się
make[4]: *** [archprepare] Błąd 2
make[4]: Opuszczenie katalogu '/home/piotrek/asuswrt-merlin/release/src-rt-7.14.114.x/src/linux/linux-2.6.36'
Makefile:805: polecenia dla obiektu 'kernel' nie powiodły się
make[3]: *** [kernel] Błąd 2
make[3]: Opuszczenie katalogu '/home/piotrek/asuswrt-merlin/release/src/router'
Makefile:193: polecenia dla obiektu 'all' nie powiodły się
make[2]: *** [all] Błąd 2
make[2]: Opuszczenie katalogu '/home/piotrek/asuswrt-merlin/release/src-rt-7.14.114.x/src'
Makefile:2569: polecenia dla obiektu 'bin' nie powiodły się
make[1]: *** [bin] Błąd 2
make[1]: Opuszczenie katalogu '/home/piotrek/asuswrt-merlin/release/src-rt-7.14.114.x/src'
Makefile:2639: polecenia dla obiektu 'rt-ac88u' nie powiodły się
make: *** [rt-ac88u] Błąd 2
root@VM-Ubuntu:/home/piotrek/asuswrt-merlin/release/src-rt-7.14.114.x/src# 

Please help me. I am begginer user linux.

RMerl, 2016-08-27:
Please use the support forums for support requests.

closed 2016-08-27 by RMerl

pepeEL, 2016-08-27:
Which forum ? Snb ?