Asuswrt-Merlin/asuswrt-merlin.legacy #1015: Unable to save System settings

opticron, 2016-08-27:
When attempting to save Administration->System settings on RT-AC66U 380.61, an alert box pops up claiming "This port is for Download Master." without indication which port setting is at fault. As it turns out, the isPortConflict validation function thinks "" is reserved for Download master and it is referring to the invisible HTTP port field. I couldn't find where this exists in the codebase after a quick search, so I'm guessing this file is dynamically generated and Download Master just doesn't have a port set (it's not actually installed as per the USB application page prompting to install it).

RMerl, 2016-08-27:
What port numbers are you entering on that page?

opticron, 2016-08-27:
The only port number entry field is for SSH and it's set to 22, but there is a non-visible field for HTTP port that is empty (based on looking at the source).

RMerl, 2016-09-03:
The webui validates https (lan and wan) and http (wan) against various services such as DM and AiCloud.  None of the three http(s) ports should ever be empty, something's wrong with your settings.  You will have to fix the nvram setting for the port that's currently empty on your setup.

opticron, 2016-09-06:
It appears this was an issue with nvram settings not being defaulted properly coming from dd-wrt (even though I cleared during flashing). A second nvram clear fixed the problem.

closed 2016-09-06 by opticron