Asuswrt-Merlin/asuswrt-merlin.legacy #1017: Traffic Monitor Wired Tab

antonlobanov, 2016-08-28:
in traffic monitor, wire tab, not drawn graph download

RMerl, 2016-08-28:
Works for everyone else, so most likely an issue on your end.  Try a different browser.

closed 2016-08-28 by RMerl

antonlobanov, 2016-08-28:
[image]
[image]

It's look like not browser issue, i try on chrome and explorer.
I running speedtest.net on screens. You may see that total received bytes on wired tab much less than in wan tab. That why i cant see graph, it is not visible because of the scale.

RMerl, 2016-08-29:
See the note about NAT acceleration on that page - it must be disabled for accurate results.