Asuswrt-Merlin/asuswrt-merlin.legacy #1021: Add a "Turn LEDs Off" option

feinstein, 2016-08-30:
The WPS button can be used as a radio toggle on my RT-AC66u, I think it would be a good addition to make it capable of turning all the LEDs Off too. 

At night my router glooms this blue light on my face and it annoys me to sleep, it would be nice to be able to turn the lights off when I want them.

nsymms, 2016-08-30:
Similar ideas have been proposed and rejected in the past. See https://github.com/RMerl/asuswrt-merlin/issues/847 and https://github.com/RMerl/asuswrt-merlin/pull/919.

feinstein, 2016-08-30:
I can see the similarity but my suggestion is a bit different. 

It's not a script that will run on scheduled times, it's a button functionality. 

This way if I want to see if the router is working or not at 03:00am I still can with a simple push of a button (which will happen a lot for me at least).

Monter, 2016-08-30:
If you want to check whether the router is simply log in to the SSH, WebGUI or send a ping. On the router it is too warm (if you already have approached and press a button), but both the generated heat as well as flashing LEDs do not provide 100% that the router is working properly.

feinstein, 2016-08-30:
Logging in takes too much time for a simple and quick status check, I rather press a button while I pass by the router. Being warm doesn't indicate it's running, only that it has power. 

So IMHO, logging it's too complicated, temperature check too simple. Lights button stays on the between, the sweet spot. Afterall that's why the lights were created for in the first place.

oldgringo, 2016-08-31:
I think it's way too complicated to hassle with a button. Cover the leds with some opaque material, it's the simplest solution. :-D
Or apply some semitransparent dark foil on the leds to make them less bright.

feinstein, 2016-08-31:
That's what I do currently, I just wanted a cleaner solution. 

I would click the button once and leave the LEDs Off all the time. Just once in a while I would click it again to turn them on to see whats going on there, like if the USB drive is connected or if the internet is up. 

I truly think it's a nice add, clean and neat.

RMerl, 2016-08-31:
I don't want to add an option to disable the LEDs because after that, someone else will ask for another option to pile up on that button.

I've decided instead that I would add the functionality to allow the button to run any user defined script.  That would allow one to make it do anything they wanted, from toggling the LED state, to triggering a reboot, stopping/starting any service they wanted, etc...

There's currently no ETA as to when this will be done.

feinstein, 2016-08-31:
Fair enough! I think this satisfies my needs and attends even more. Thanks!

b3nk3, 2016-09-09:
That is an awesome idea @RMerl. Until then I'll be covering the led's with tape for my guests :D

BAGELreflex, 2016-10-17:
@b3nk3 @feinstein If having the LED's disabled entirely (as seems to be the case for b3nk3 atleast) then that functionality already exists. You can do that in the WebUI by browsing to `Tools > Other Settings > Miscellaneous Options > Stealth Mode`. This can also be toggled via ssh by issuing:
```
nvram set led_disable=1
nvram commit
```

feinstein, 2016-11-24:
@BAGELreflex sorry for the delay, I confess I forgot to check my messages.

Is this command possible with the factory firmware? I have a new unit and I am looking forward to mod it just yet.

BAGELreflex, 2016-11-25:
@feinstein I do not currently have a device flashed with Asus' stock firmware, so I cannot verify. However I'm fairly certain that **Stealth Mode** is part of the Merlin build, as there is no mention in the stock FW documentation: http://dlcdnet.asus.com/pub/ASUS/wireless/RT-AC66U/E7891_RT_AC66U_Manual.pdf

feinstein, 2016-11-28:
@BAGELreflex I was referring more to a script or a command that could be submitted to the stock firmware. I am pretty sure as well there isn't an option for a stealth mode on the stock firmware.

Are there any channels were users can submit ideas to the stock firmware developers?

orangy, 2017-06-21:
+1 to a script attached to a WPS button. I would use it force switch between WANs in a Dual WAN setting.

toxuin, 2018-01-09:
It would be pretty nice to have that button toggle a OpenVPN client connection and then change a LED state when connected. Give us the user script and we'll write the rest.

bits3rpent, 2018-01-09:
Is there not a stealth mode that accomplishes that already?

toxuin, 2018-01-09:
Well, that's not quite what I meant. Stealth mode would disable LEDs, where I would like LEDs change state on network connection events – controlled from script. And a button that starts a user script.