Asuswrt-Merlin/asuswrt-merlin.legacy #1023: Increase the max. elements for ipset from 65K to 1 million.

blackfuel, 2016-09-01:
Increase the max. elements for **ipset** from 65K to 1 million.
Search for:   #define MAX_RANGE 0x0000FFFF
Replace with: #define MAX_RANGE 0x000FFFFF
https://github.com/RMerl/asuswrt-merlin/search?utf8=%E2%9C%93&q=%23define+MAX_RANGE+0x0000FFFF

The file is ip_set.h, but there is more than one occurance of this file.

I'm running with this on my RT-AC68U router now.

RMerl, 2016-09-02:
Won't having such a large limitation negatively impact performance and increase memory usage for all other usual usage scenarios?

RMerl, 2016-09-05:
Changing MAX_RANGE is a bad idea.  I see portions of the code where they use it to determine the highest allowed port number (!).  MAX_PORT is set to MAX_RANGE+1 in kernel/include/linux/netfilter_ipv4/ip_set_getport.h .

I'm considering upgrading the ARM platform to ipset 6 at some point, I'm still debating the potential impact it could have to support different ipset versions on MIPS and ARM.  As the MIPS platform is going to be gradually deprecated, this might become less of an issue.

RMerl, 2016-10-17:
ARM has been moved to ipset 6.x.  How does that version look like regarding the maximum number of elements allowed?

blackfuel, 2016-10-18:
On my RT-AC68U with Asuswrt-Merlin 380.63_alpha1 (plus my own kernel mods), the ipset 6.x program appears to be working correctly.  All I had to do was update the header inside my saved ipsets so it correctly restores a saved set using the new _maxelem_ option.

However, iptables does not appear to recognize any ipsets.

**# ipset create testset hash:ip**

**# iptables -A INPUT -i eth0 -m set --match-set testset src -j ACCEPT**
_iptables: No chain/target/match by that name._

**# iptables -A INPUT -i eth0 -p tcp -j SET --add-set testset src**
_iptables: No chain/target/match by that name._

I'm doing other kernel mods too, so hopefully I haven't broken something.

RMerl, 2016-10-18:
I only tested basic stuff against dnsmasq, I didn't test iptables interaction.  I can reproduce the problem here on my own RT-AC88U as well.  I'll have to see if the merge might have been missing some iptables userspace patching.  I'll close this issue, and open a separate one for the iptables issues.

closed 2016-10-18 by RMerl