Asuswrt-Merlin/asuswrt-merlin.legacy #1024: PPTP only works for first connection.

CKret, 2016-09-02:
When enabling the PPTP server and connecting to it VPN works as expected.
If disconnecting and trying to reconnect then a connection cannot be established.
Disabling PPTP and enabling it again will allow the user to connect again only once before a disable/enable is required.

RMerl, 2016-09-05:
Can't reproduce the issue here, I was able to repeatedly disconnect and reconnect to the PPTPD server.

CKret, 2016-09-05:
Forgot to mention: on an AC-66U router.
I've had this problem since... well... a couple of years ago...
Asus firmware or Merlin... Same problem....
No big deal since there is a way around the problem but still an annoyance...

RMerl, 2016-09-26:
Can't reproduce and never had any other report of such a behaviour, so I suspect it's related to the client.

closed 2016-09-26 by RMerl